Stefal/rtkbase #326: Ublox M8t Orange pi (Status problems)

hdtopo, 2023-08-14:
Amigos estou com um problema quem pode me ajudar fico grato!
Estou usando Orange pi com módulo Ublox M8T e RtkBase, as configurações estão corretas como porta esta tudo ok, consigo criar o arquivo.ubx normalmente, porem estou com problema na pagina STATUS , não consigo receber as coordenadas e nem ver as barras de satélites, meu problema é somente na visualização dessa página Status o restante esta funcionando normalmente. algum pode me ajudar?
[image]
[image]
[image]

jaerabr, 2023-08-26:
Imagino que não tenha mas suporte nativo para o m8t nas versoes mais novas do RtkBase use uma versão mais antiga tipo a Rtkbase 2.2.0 
[image]
[image]

jaerabr, 2023-08-26:
Na rpi zero funciona na 2.4.1

[image]

closed 2023-12-28 by Stefal