Stefal/rtkbase #330: Web Server not running

Davidwedel, 2023-10-13:
Orange Pi zero3. OS provided in this repo would never boot up. Armbian doesn't have support either. So I used Ubuntu from [here](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-Zero-3.html). I do not know if this is an RTKBase issue or an OS error.

Commands and their outputs:
orangepi@orangepizero3:~$ sudo systemctl status rtkbase_web
● rtkbase_web.service - RTKBase Web Server
Loaded: loaded (/etc/systemd/system/rtkbase_web.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Fri 2023-10-13 12:59:14 UTC; 23s ago
Process: 4125 ExecStart=/usr/bin/python3 /home/orangepi/rtkbase/web_app/server.py (code=exited, status=1/FAILURE)
Main PID: 4125 (code=exited, status=1/FAILURE)
CPU: 3.811s

orangepi@orangepizero3:~$ sudo journalctl -u rtkbase_web
Oct 13 12:56:54 orangepizero3 systemd[1]: Started RTKBase Web Server.
Oct 13 12:56:57 orangepizero3 python3[4029]: /usr/local/lib/python3.10/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
Oct 13 12:56:57 orangepizero3 python3[4029]:   warnings.warn("Setuptools is replacing distutils.")
Oct 13 12:56:57 orangepizero3 python3[4029]: Traceback (most recent call last):
Oct 13 12:56:57 orangepizero3 python3[4029]:   File "/home/orangepi/rtkbase/web_app/server.py", line 59, in <module>
Oct 13 12:56:57 orangepizero3 python3[4029]:     from flask_bootstrap import Bootstrap4
Oct 13 12:56:57 orangepizero3 python3[4029]:   File "/usr/local/lib/python3.10/dist-packages/flask_bootstrap/__init__.py", line 3, in <module>
Oct 13 12:56:57 orangepizero3 python3[4029]:     from flask import current_app, Markup, Blueprint, url_for
Oct 13 12:56:57 orangepizero3 python3[4029]:   File "/usr/local/lib/python3.10/dist-packages/flask/__init__.py", line 5, in <module>
Oct 13 12:56:57 orangepizero3 python3[4029]:     from .app import Flask as Flask
Oct 13 12:56:57 orangepizero3 python3[4029]:   File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 30, in <module>
Oct 13 12:56:57 orangepizero3 python3[4029]:     from werkzeug.urls import url_quote
Oct 13 12:56:57 orangepizero3 python3[4029]: ImportError: cannot import name 'url_quote' from 'werkzeug.urls' (/usr/local/lib/python3.10/dist-packages/werkzeug/urls.py)
Oct 13 12:56:58 orangepizero3 systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=1/FAILURE
Oct 13 12:56:58 orangepizero3 systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
Oct 13 12:56:58 orangepizero3 systemd[1]: rtkbase_web.service: Consumed 4.057s CPU time.
Oct 13 12:57:28 orangepizero3 systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 1.
Oct 13 12:57:28 orangepizero3 systemd[1]: Stopped RTKBase Web Server.
Oct 13 12:57:28 orangepizero3 systemd[1]: rtkbase_web.service: Consumed 4.057s CPU time.
Oct 13 12:57:28 orangepizero3 systemd[1]: Started RTKBase Web Server.
Oct 13 12:57:31 orangepizero3 python3[4099]: /usr/local/lib/python3.10/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
Oct 13 12:57:31 orangepizero3 python3[4099]:   warnings.warn("Setuptools is replacing distutils.")
Oct 13 12:57:31 orangepizero3 python3[4099]: Traceback (most recent call last):
Oct 13 12:57:31 orangepizero3 python3[4099]:   File "/home/orangepi/rtkbase/web_app/server.py", line 59, in <module>
Oct 13 12:57:31 orangepizero3 python3[4099]:     from flask_bootstrap import Bootstrap4
Oct 13 12:57:31 orangepizero3 python3[4099]:   File "/usr/local/lib/python3.10/dist-packages/flask_bootstrap/__init__.py", line 3, in <module>

If you need more info, let me know.

Davidwedel, 2023-10-13:
https://github.com/pallets/werkzeug/pull/2768
i suspect this is the issue, but I could easily be wrong

torriem, 2023-10-20:
Werkzeug needs to be pinned to version 2.2.2.  Pip installs the latest version 3.0.0 by default. The solution is to add `Werkzeug==2.2.2` to requirements.txt

Stefal, 2023-10-21:
@torriem Good catch ! Thank you !

torriem, 2023-10-22:
I submitted a pull request that adds that requirement.

Stefal, 2023-11-02:
Fixed with https://github.com/Stefal/rtkbase/releases/tag/2.4.2

closed 2023-11-02 by Stefal