Stefal/rtkbase #339: Ntrip A Service will not stay on.

carkah, 2023-11-03:
**Describe the bug**
I have Main Service and Ntrip A Service configured correctly on toggled on.  I go to the status page and when I go back to settings Ntrip A service is toggled off but Main service is still on.  It actually does it for any of the toggles now that I have tried some more

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings
2. Click on to toggle for Ntrip service A
3.  Click on status
4. click on settings
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.2.2
 - OS: Debian 11 (Bullseye)
 - SBC Xunlong Orange Pi Zero
 - GNSS Device: U-blox_ZED-F9P - 1.11
 - Connection type: usb

Stefal, 2023-11-04:
Hi!
Please copy paste here what you see in the diagnostic page about str2str_tcp and str2str_ntrip_A

carkah, 2023-11-04:
I think this is what you are looking for.  let me know if you need something else

str2str_tcp.service : Active
● str2str_tcp.service - RTKBase Tcp
Loaded: loaded (/etc/systemd/system/str2str_tcp.service; enabled; vendor preset: enabled)
Active: active (running) since Sat 2023-11-04 04:11:50 UTC; 9h ago
Process: 1143 ExecStart=/home/basegnss/rtkbase/run_cast.sh in_serial out_tcp (code=exited, status=0/SUCCESS)
Main PID: 1150 (str2str)
Tasks: 2 (limit: 906)
Memory: 352.0K
CPU: 5min 24.234s
CGroup: /system.slice/str2str_tcp.service
└─1150 /usr/local/bin/str2str -in serial://ttyACM0:115200:8:n:1#ubx -out tcpsvr://:5015 -b 1 -t 0 -fl /home/basegnss/rtkbase/logs/str2str_tcp.log

Nov 04 14:04:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 14:04:08 basegnss run_cast.sh[1150]: 2023/11/04 14:04:27 [CC---] 160344392 B 32908 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 14:05:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 14:05:08 basegnss run_cast.sh[1150]: 2023/11/04 14:05:27 [CC---] 160579992 B 32892 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 14:06:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 14:06:08 basegnss run_cast.sh[1150]: 2023/11/04 14:06:27 [CC---] 160815592 B 31936 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 14:07:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 14:07:08 basegnss run_cast.sh[1150]: 2023/11/04 14:07:27 [CC---] 161051192 B 31904 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 14:08:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 14:08:08 basegnss run_cast.sh[1150]: 2023/11/04 14:08:27 [CC---] 161288248 B 32159 bps (0) /dev/ttyACM0 (1) 2 clients

-- Journal begins at Sat 2023-11-04 04:11:30 UTC, ends at Sat 2023-11-04 14:08:36 UTC. --
Nov 04 04:11:30 basegnss run_cast.sh[570]: stream server start
Nov 04 13:05:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:05:08 basegnss run_cast.sh[1150]: 2023/11/04 13:05:26 [CC---] 146132236 B 36973 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:06:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:06:08 basegnss run_cast.sh[1150]: 2023/11/04 13:06:26 [CC---] 146402400 B 37288 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:07:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:07:08 basegnss run_cast.sh[1150]: 2023/11/04 13:07:26 [CC---] 146674296 B 36792 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:08:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:08:08 basegnss run_cast.sh[1150]: 2023/11/04 13:08:26 [CC---] 146943768 B 36699 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:09:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:09:08 basegnss run_cast.sh[1150]: 2023/11/04 13:09:26 [CC---] 147212052 B 36619 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:10:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:10:08 basegnss run_cast.sh[1150]: 2023/11/04 13:10:26 [CC---] 147480268 B 36808 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:11:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:11:08 basegnss run_cast.sh[1150]: 2023/11/04 13:11:27 [CC---] 147747140 B 35802 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:12:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:12:08 basegnss run_cast.sh[1150]: 2023/11/04 13:12:27 [CC---] 148014548 B 35952 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:13:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:13:08 basegnss run_cast.sh[1150]: 2023/11/04 13:13:27 [CC---] 148279844 B 37016 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:14:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:14:08 basegnss run_cast.sh[1150]: 2023/11/04 13:14:27 [CC---] 148544444 B 36416 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:15:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:15:08 basegnss run_cast.sh[1150]: 2023/11/04 13:15:27 [CC---] 148805900 B 35244 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:53:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:53:08 basegnss run_cast.sh[1150]: 2023/11/04 13:53:27 [CC---] 157856704 B 30309 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:54:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:54:08 basegnss run_cast.sh[1150]: 2023/11/04 13:54:27 [CC---] 158078428 B 30096 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:55:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:55:08 basegnss run_cast.sh[1150]: 2023/11/04 13:55:27 [CC---] 158301164 B 30561 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:56:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:56:08 basegnss run_cast.sh[1150]: 2023/11/04 13:56:27 [CC---] 158526064 B 30594 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Nov 04 13:57:08 basegnss systemd-journald[467]: Suppressed 11 messages from str2str_tcp.service
Nov 04 13:57:08 basegnss run_cast.sh[1150]: 2023/11/04 13:57:27 [CC---] 158751072 B 30640 bps (0) /dev/ttyACM0 (1) 127.0.0.1
[... cut ...]
Nov 04 04:11:31 basegnss run_cast.sh[575]: -o e n u antenna offset (e,n,u) (m)
Nov 04 04:11:31 basegnss run_cast.sh[575]: -l local_dir ftp/http local directory []
Nov 04 04:11:31 basegnss run_cast.sh[575]: -x proxy_addr http/ntrip proxy address [no]
Nov 04 04:11:31 basegnss run_cast.sh[575]: -b str_no relay back messages from output str to input str [no]
Nov 04 04:11:31 basegnss run_cast.sh[575]: -t level trace level [0]
Nov 04 04:11:31 basegnss run_cast.sh[575]: -fl file log file [str2str.trace]
Nov 04 04:11:31 basegnss run_cast.sh[575]: -h print help

carkah, 2023-11-04:
[image]
[image]

Stefal, 2023-11-05:
The Ntrip service refuse your parameters but I don't know why. 
 - Do you use only ASCII char for your mount name (without space or special char)?
 - Are the coordinates 'lat long ele' with only digits, dot, and '-' if needed ?

carkah, 2023-11-05:
ahhh I did not have an elevation entered in!  Thank You.  I will for sure sponsor you for the help.  Is elevation in feet, meters, or other?

Stefal, 2023-11-05:
> ahhh I did not have an elevation entered in! Thank You. I will for sure sponsor you for the help. Is elevation in feet, meters, or other?

in meters (SI) please.

closed 2023-11-12 by Stefal