Stefal/rtkbase #343: Right RAW messages types for RTKbase

Jaxz13, 2023-11-16:
I try to make base station based on RTK3B [https://www.ardusimple.com/product/simplertk3b-x5/](url) and RTKbase.

I get messages from RTK3B to raspberry uart1, /dev/ttyS0. Main sevice stays green in RTKbase, but nothing happens in status page - no satellite bars or location information at all.

What are right messages types for RTKbase?
[image]

What is right receiver format in main settings?
[image]

jaerabr, 2023-11-16:
Hello, your gnss module is sending rtcm3 to rtkase, so change ubx to rtcm3 in main service

[image]

Stefal, 2023-11-16:
@jaerabr is right

I'm working on a new RTKBase release which should be able to use the Mosaic X5 with raw messages. It will be available next year.

Jaxz13, 2023-11-16:
> @jaerabr is right
> 
> I'm working on a new RTKBase release which should be able to use the Mosaic X5 with raw messages. It will be available next year.

Sounds good that Mosaic integration.. 

Actually I need (or my customer needs) base station with two differrent base coordinates height setting -> and two different NTRIP services (same Lat and Lon coordinates, different height). My thinking is to do one NTRIP service in mosaic X5 and other in RTKbase.

If that setting, as described above is possible to implement in RTKbase, I will be happy.

Stefal, 2023-11-16:
> Actually I need (or my customer needs) base station with two differrent base coordinates height setting -> and two different NTRIP services (same Lat and Lon coordinates, different height). 
> If that setting, as described above is possible to implement in RTKbase, I will be happy.

Different height for the 2 ntrip services inside RTKBase is not possible. 

But you can hack RTKBase :
 - add a new "position" value in settings.conf, ie "position_B"
 - edit `out_caster_B="ntrips://:${svr_pwd_b}@${svr_addr_b}:${svr_port_b}/${mnt_name_b}#rtcm3 -msg ${rtcm_msg_b} -p ${position}"` to use this new variable :  `out_caster_B="ntrips://:${svr_pwd_b}@${svr_addr_b}:${svr_port_b}/${mnt_name_b}#rtcm3 -msg ${rtcm_msg_b} -p ${position_B}"`

And it should work 😄

Jaxz13, 2023-11-16:
Now there is beams.. but still no position. Waiting more than 10 minutes that position after startup. Position seen right in Mosaic setting page.

[image]

Jaxz13, 2023-11-16:
> > Actually I need (or my customer needs) base station with two differrent base coordinates height setting -> and two different NTRIP services (same Lat and Lon coordinates, different height).
> > If that setting, as described above is possible to implement in RTKbase, I will be happy.
> 
> Different height for the 2 ntrip services inside RTKBase is not possible.
> 
> But you can hack RTKBase :
> 
>     * add a new "position" value in settings.conf, ie "position_B"
> 
>     * inside run_cast_.sh edit `out_caster_B="ntrips://:${svr_pwd_b}@${svr_addr_b}:${svr_port_b}/${mnt_name_b}#rtcm3 -msg ${rtcm_msg_b} -p ${position}"` to use this new variable :  `out_caster_B="ntrips://:${svr_pwd_b}@${svr_addr_b}:${svr_port_b}/${mnt_name_b}#rtcm3 -msg ${rtcm_msg_b} -p ${position_B}"`
> 
> 
> And it should work 😄

Thanks for the info! I will try that.

Jaxz13, 2023-11-21:
> Now there is beams.. but still no position. Waiting more than 10 minutes that position after startup. Position seen right in Mosaic setting page.
> [image]

Still no position in status page.. What are the right settings in Mosaic-x5?
[image]

RTKBase gets position from mosaic but setting in RTKbase - Main service - base coordinates, does not effect anything in NTRIP services. Ntrip service works but Base station position is same what position is seen in mosaic-X5.

jaerabr, 2023-11-21:
Good morning, do a test, disable the rtcm messages that are sent to the ECEF location in mosaic-X5, example: 1005, 1006 and 1032, I do this with the unicore um980 module and it works

Jaxz13, 2023-11-21:
> Good morning, do a test, disable the rtcm messages that are sent to the ECEF location in mosaic-X5, example: 1005, 1006 and 1032, I do this with the unicore um980 module and it works

Thanks for the reply... Tried to disable those mentioned messages. Still not working.

What RTCM3 messages sends your unicore um980 to rtkbase?

jaerabr, 2023-11-21:
> > Bom dia, faça um teste, desative as mensagens rtcm que são enviadas para o local ECEF no mosaico-X5, exemplo: 1005, 1006 e 1032, faço isso com o módulo unicore um980 e funciona
> 
> Obrigado pela resposta... Tentei desativar as mensagens mencionadas. Ainda não funciona.
> 
> Quais mensagens RTCM3 enviam seu unicore um980 para rtkbase?

I leave the 1019, 1020, 1042, 1046, 1077, 1087, 1097 and 1124 active, all at 1hz

Jaxz13, 2023-11-22:
> I leave the 1019, 1020, 1042, 1046, 1077, 1087, 1097 and 1124 active, all at 1hz

It needed those  1019, 1020, 1042, 1046.

Stefal, 2024-01-31:
Could we close this issue?

closed 2024-03-24 by Stefal