Stefal/rtkbase #348: "Create Rinex file" in "Convert raw file archive (zip) to rinex file" is disabled

bendauphinee, 2023-12-21:
**Describe the bug**
On a fresh install, I'm unable to use the convert functionality.

**To Reproduce**
Steps to reproduce the behavior:
- Install
- Configure GPS
- Enable File service, don't change any settings
- Open Logs
- Click "Convert to Rinex"

**Expected behavior**
The button works, and I can convert.

**Software/Hardware (please complete the following information):**
| | |
|---|---|
| RTKBase | v2.4.2 |
| GNSS | U-blox_ZED-F9P - 1.32 |
| Board | Raspberry Pi 3 Model B Rev 1.2 |
| OS | Debian 11 (Bullseye) 64 bit |
| Storage | 25.48GB available of 31.07GB |
| Connection | USB |

**Screenshots**
[image]

Stefal, 2023-12-21:
Did you try to convert a ubx file or a zip file ? It works only with zip file.

bendauphinee, 2023-12-21:
Ok so this must be me misunderstanding this functionality then. I initially installed this because one of the main feature is listed as "Convert raw data to Rinex", and there's a convert button next to every file listed.

Do I need to somehow zip the files that it's logging before I run the converter, or is there some other process to get a file converted out of this?

jaerabr, 2023-12-21:
Good morning, taking advantage of this topic, would you like to modify this function to convert ubx or rtcm3 to rinex? Could anyone tell me where I can find information about this change?

Stefal, 2023-12-21:
The main goal of rinex conversion is to create a file to compute the accurate base station position. To do this, we usually need a 24h log file.

So each night, the data from the previous day is archived in a zip file. Then you could convert it to a rinex file.
If you want to convert to rinex data from the same day, you can download the ubx file and convert it to rinex on another computer.

bendauphinee, 2023-12-21:
Ok, if that's the case (since we can get positioning with as little as 6 hours of data), then the convert button should be disabled, with a hover text that indicates why. Alternatively or additionally, some documentation hint on the page that says that, and maybe an update to the main page for this site.

Or even cooler, if it's not a massive trouble, can we get convert enabled for raw files?

Thanks for the answer to this!

Stefal, 2023-12-21:
> Good morning, taking advantage of this topic, would you like to modify this function to convert ubx or rtcm3 to rinex? Could anyone tell me where I can find information about this change?

Why not !
If you want to add this feature, you need to edit https://github.com/Stefal/rtkbase/blob/dev/tools/convbin.sh then create a pull request on the dev branch.

Stefal, 2023-12-21:
> some documentation hint on the page that says that, and maybe an update to the main page for this site.

You're right, RTKBase need a better documentation. I've wrote a partial one in french (https://github.com/CentipedeRTK/docs-centipedeRTK/blob/master/docs/base/Guide_RTKBase.md) but it needs an english/.../... one.

commit c303ffcb (2024-12-17): Convert to rinex for non-zipped files.
Fix #348
closed 2025-02-16 by Stefal