Stefal/rtkbase #356: Install fails on Debian/Raspian 12 Bookworm

kinghaigy, 2024-01-16:
**Describe the bug**
The installation script fails to build pystemd. The detailed output from the script at the point of failure is:

```
Building wheels for collected packages: pystemd
  Building wheel for pystemd (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [41 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/__version__.py -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/__init__.py -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/exceptions.py -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/utils.py -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/run.py -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/base.py -> build/lib.linux-aarch64-cpython-311/pystemd
      creating build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      copying pystemd/systemd1/__init__.py -> build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      copying pystemd/systemd1/manager.py -> build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      copying pystemd/systemd1/unit_signatures.py -> build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      copying pystemd/systemd1/unit.py -> build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      creating build/lib.linux-aarch64-cpython-311/pystemd/machine1
      copying pystemd/machine1/__init__.py -> build/lib.linux-aarch64-cpython-311/pystemd/machine1
      copying pystemd/machine1/machine.py -> build/lib.linux-aarch64-cpython-311/pystemd/machine1
      copying pystemd/machine1/manager.py -> build/lib.linux-aarch64-cpython-311/pystemd/machine1
      creating build/lib.linux-aarch64-cpython-311/pystemd/DBus
      copying pystemd/DBus/__init__.py -> build/lib.linux-aarch64-cpython-311/pystemd/DBus
      copying pystemd/dbuslib.pyi -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/__init__.pyi -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/base.pyi -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/utils.pyi -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/daemon.pyi -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/RELEASE -> build/lib.linux-aarch64-cpython-311/pystemd
      copying pystemd/systemd1/unit.pyi -> build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      copying pystemd/systemd1/manager.pyi -> build/lib.linux-aarch64-cpython-311/pystemd/systemd1
      running build_ext
      building 'pystemd.dbusexc' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/pystemd
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c pystemd/dbusexc.c -o build/temp.linux-aarch64-cpython-311/pystemd/dbusexc.o
      pystemd/dbusexc.c:213:12: fatal error: longintrepr.h: No such file or directory
        213 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pystemd
  Running setup.py clean for pystemd
Failed to build pystemd
ERROR: Could not build wheels for pystemd, which is required to install pyproject.toml-based projects
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run the install script on a Debian 12 based system

**Expected behavior**
The install works as it did in Debian 11 and runs RTKBase as designed. 

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.4.2
 - OS: Raspbian 12 x64
 - SBC Raspberry Pi 4B
 - GNSS Device: Ublox F9P from Ardusimple Board
 - Connection type: USB

**Additional context**
This is the same issue as number 342 (https://github.com/Stefal/rtkbase/issues/342) but it's going to pop up more. I'm mostly interested in if you've got a fix in the pipeline coming soon as I don't want to downgrade this pi if I can help it. This is a fresh install of Raspbian 12.

Stefal, 2024-01-16:
Hi!

I'm working on the next release, with Bookworm compatibility.
If you want you can install the dev release, which need testing : 

```
wget https://raw.githubusercontent.com/Stefal/rtkbase/dev/tools/install.sh -O install.sh
chmod +x install.sh
sudo ./install.sh --all repo --rtkbase-repo dev
```

kinghaigy, 2024-01-17:
Great work! I'm happy to report the installation passed with no errors and that was after multiple failed installs with the main branch. I'll let you know here if there are any functional issues once I've had time to play with it.

kinghaigy, 2024-01-17:
I'm back quite quickly. There's nothing to say anything if failing in the install script but on the status page, a pop-up saying "Main service is off or malfunctioning. Nothing could be displayed here." shows up. All services appear to launch fine (nothing complains - is there something I can check for you here?)

The settings page says the GNSS reciever is 'unknown' despite the install script finding and configuring (apparently) correctly:

```
################################                                                                          
USB GNSS RECEIVER DETECTION                                                                               
################################                                                                          
################################                                                                          
UART GNSS RECEIVER DETECTION                                                                              
################################                                                                          
/dev/ttyGNSS  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver -  115200                                
GNSS RECEIVER DETECTED: /dev/ttyGNSS  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver  -  115200       
################################                                                                          
CONFIGURE GNSS RECEIVER                                                                                   
################################                                                                          
U-Blox ZED-F9P detected                                                                                   
Resetting ZED-F9P to default settings                                                                     
UBX-ACK-ACK:                                                                                              
  ACK to Class x06 (CFG) ID x09 (CFG)                                                                     
                                                                                                          
Set UART Baudrate....                                                                                     
UBX-ACK-ACK:                                                                                              
  ACK to Class x06 (CFG) ID x8a (VALSET)                                                                  
                                                                                                          
Sending settings....                                                                                      
GNSS Configuration: done                                                                                  
################################                                                                          
STARTING SERVICES                                                                                         
################################                                                                          
################################                                                                          
END OF INSTALLATION                                                                                       
You can open your browser to http://10.0.0.30 172.17.0.1                                                  
################################                                                                          
```

Stefal, 2024-01-18:
> The settings page says the GNSS reciever is 'unknown' despite the install script finding and configuring (apparently) correctly

Hmm, strange, I'll try to repro.
Did you check if the values are correct in settings.conf ?

Stefal, 2024-01-18:
> I might be in luck. The config looked good. I checked /dev/ for ttyGNSS and it was missing for some reason. I did check output from the serial port and could see stuff coming in last night so it was definitely there when I wrote my last post but I rebooted ther raspberry pi and it seems to have found it and is working as I think it should be. I've just got to set my base position up now.

Is ttyGNSS still missing ? 

> Something I wasn't sure from reading the install setup - does your installation script set the GPS up to have a fixed location that I set through the UI? Or is that something I do through ublox control center? 

The script doesn't set a fixed location inside the Gnss receiver, because it is not used at all. The settings that are sent are [here](https://github.com/Stefal/rtkbase/blob/dev/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg)
You don't have to set anything in U-Center.

> At the moment, the coordiantes listed under the signal strengths are not the same as the base coordinates I have entered in the settings page.

The coordinates you see on the status page is a realtime position calculated with RTKLib. There is always a small discrepancy between the real time position and the calculated position.

kinghaigy, 2024-01-18:
ttyGNSS re-appeared after restarting. No idea why it went missing. 

The GPS needs to know its position in order for it to be accurate (relative to my surveyed coordinates) for RTK. Do the base coordinate settings on the settings page get added to the RTCM3 message? I don't know the message structure well enough to know what goes where but I thought the coordinates had to be set through u-center in TMODE3. That information came from loosely following a Ardusimple RTK guide.

It's offtopic, but I appreciate your help.

Stefal, 2024-01-18:
> Do the base coordinate settings on the settings page get added to the RTCM3 message?

Exactly !
And RTKBase generates the Rtcm messages by itself, it doesn't use the one the F9P can create.

kinghaigy, 2024-01-18:
That's cool! In that case, that might be worth noting somewhere in README.md as I was under the impression that RTKBase just exposed multiple avenues to get the RTCM data from the GPS its connected to. Good to know.

Well done making something so easy to get up and running! I'll let you know over the next few days if I have other issues. Cheers!

Stefal, 2024-01-31:
Fixed with https://github.com/Stefal/rtkbase/releases/tag/v2.5.0