Stefal/rtkbase #357: Easy installation not working RPi 0W 32bit lite

karlranseierausrom, 2024-01-23:
**Describe the bug**
Easy installation on a fresh installation RPi 0W 32 bit is not working.

**To Reproduce**
Steps to reproduce the behavior:
1. use imager to install 32bit lite installation
2. connect GNSS receiver and plug in sd card
3. connect via putty
4. enter install script (added sudo before chmod)
5. Error: 
```bash
sudo ./install.sh --detect-usb-gnss --configure-gnss
user for RTKBase is:  Pi
################################
GNSS RECEIVER DETECTION
################################
/dev/ttyACM0  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver
GNSS RECEIVER DETECTED: /dev/ttyACM0  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver
################################
CONFIGURE GNSS RECEIVER
################################
ubxtool: failed to import pyserial
No Gnss receiver has been set. We can't configure
```

**Expected behavior**
Succeeded installation and a running web interface

**Software/Hardware (please complete the following information):**
 - RTKBase release number X.X.X
 - OS: Raspberry Pi Os 32 bit lite
 - SBC  Raspberry  Pi Zero W
 - GNSS Device: Ublox F9P
 - Connection type: usb

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I tried

```bash
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
```
but it did not work.

best regards
karlranseier

Stefal, 2024-01-23:
Did you try `sudo ./install.sh --all` ?

BTW : You should be able to use the ready to flash image.

geofis, 2024-01-23:
Hello.

Maybe related with the Raspberry Pi OS version. @karlranseierausrom Have you tried with the Debian Bullseye distribution (Legacy, 32-bit)?

karlranseierausrom, 2024-01-24:
> sudo ./install.sh --all

Pi@RTKBase:~ $ sudo ./install.sh --all
install: option '--all' requires an argument
Try 'install.sh --help' for more information

Does not work.

karlranseierausrom, 2024-01-24:
> Hello.
> 
> Maybe related with the Raspberry Pi OS version. @karlranseierausrom Have you tried with the Debian Bullseye distribution (Legacy, 32-bit)?

I tried that first, but it did not work.

karlranseierausrom, 2024-01-24:
> BTW : You should be able to use the ready to flash image

I tried that first, but it did not work.

Stefal, 2024-01-24:
> > sudo ./install.sh --all
> 
> Pi@RTKBase:~ $ sudo ./install.sh --all install: option '--all' requires an argument Try 'install.sh --help' for more information
> 
> Does not work.

Sorry, I forgot the release argument, as written in the readme
[image]

`sudo ./install.sh --all release`

> I tried that first, but it did not work.

strange...

karlranseierausrom, 2024-01-24:
Yes, that's what I meant with easy installation. I tried that, but ir could not change the rights with chmod. The small difference was, that I needed to use
```bash
cd ~
wget https://raw.githubusercontent.com/Stefal/rtkbase/master/tools/install.sh -O install.sh
sudo chmod +x install.sh
sudo ./install.sh --all release
```

Stefal, 2024-01-24:
Strange.

@jancelin did you try your image with a Pi zero recently ?

karlranseierausrom, 2024-01-24:
Shame on me, I have selected the wrong RPi. It looks like I have selected Pi Zero 2, otherwise I could not explain, why 64 bit was offered. @geofis , so I did it now and I thought I did it before. Your comment made me to check that.

Sorry for stealing your time and thanks for your help!

closed 2024-01-27 by Stefal