Stefal/rtkbase #360: Disable File Service if free storage is less than 1Gb

gunicsba, 2024-01-25:
**Is your feature request related to a problem? Please describe.**
Today I found an RTKBase with 0% storage left. It caused no problems.

[image]

**Describe the solution you'd like**
Run a daily check to see how much free storage we have.
If it's under 1GB automatically disable File Service.

**Describe alternatives you've considered**
Pi had only a 16GB SD card and 60 days of retention. I believe that 60 days is actually too much for the 16GB SD card.
Changing the retention period to 20 days might also eliminate this problem.

Stefal, 2024-01-27:
- Another solution is to automaticaly reduce the retention period if the remaining available space is lower than x%
- 60 days of logs should use about 10GB.
- The ubx files should have been compressed in zip file, it's strange the your old files are not.

Stefal, 2024-01-27:
I forgot that there is already a protection against storage full inside https://github.com/Stefal/rtkbase/blob/master/archive_and_clean.sh :
```
#Check if there is enough available space (default value is more than 500MB), and delete oldest archive if needed.
while [[ $(check_space) -lt ${min_free_space} ]]
do
  file_to_delete=$(find . -maxdepth 1 -type f | sort -r | tail -1)
  #Test if there is a file to delete
  if [[ -z ${file_to_delete} ]]
  then
    break
  fi
  echo 'Not enough remaining space. Deleting' ${file_to_delete}
  #Exit if can't delete
  if ! $(rm -f ${file_to_delete})
  then
    break
  fi
done
```

Please check that rtkbase_archive.service and rtkbase_archive.timer are enabled.

Stefal, 2024-02-23:
Could we close this issue?

closed 2024-03-24 by Stefal