Stefal/rtkbase #363: Base Coordinates showing invalid after 2.5.0 update

carkah, 2024-02-02:
**Describe the bug**
After I updated to 2.5.0 my base coordinates show invalid and wont let me save any changes made to any of the other categories, however, base is still showing up correctly on rtk2go and on the status page.  I have 1 space in between lat and long and long and elevation

**To Reproduce**
Steps to reproduce the behavior:
1. update to 2.5.0 and error appeared.

**Software/Hardware (please complete the following information):**
 - RTKBase release number X.X.X
 - OS: Debian 11 (Bullseye)
 - SBC Xunlong Orange Pi Zero
 - GNSS Device: U-blox_ZED-F9P - 1.11
 - Connection type: usb

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Stefal, 2024-02-02:
Do you have at least two digits for your elevation value ?

carkah, 2024-02-02:
I did not, however, I tried 7.0 and 11 and neither were accepted

Stefal, 2024-02-02:
> I did not, however, I tried 7.0 and 11 and neither were accepted

did you try `7.00` ? (it's two digits after the comma)

carkah, 2024-02-03:
Ah yes that is it.  Thank you

closed 2024-02-03 by carkah