Stefal/rtkbase #370: Ublox F9P Survey Mode in for temporary solution

DavideLentini, 2024-02-28:
I find the idea of this project fantastic , but perhaps far from my goal . 
I would like to make a mobile base and not fixed , and I would then need to start a Survey with Min accuracy and Min time , how exactly does the Survey In mode of the Ublox F9P work . Is this project able to do this ? Thank you very much

jaerabr, 2024-03-01:
Hello, I use RTKbase as a base GNSS kit and rover with UHF radio and it works perfectly, please provide more details about your objective.
https://photos.app.goo.gl/UihiRzKDiyK1R9LF8

DavideLentini, 2024-03-01:
@jaerabr Thanks ! 
My goal is to use a Raspberry Pi with RTKBase software , as a mobile base , to inject rtk correction messages into my local network , I would probably use TCP method instead of Ntrip .
Now from what I understand this library was developed as a FIXED system , right ? and not as a mobile .
So I would need , arrived on a new place , to turn on the raspberry Pi , to which is connected via USB an Ublox F9P devkit : 
https://www.u-blox.com/en/product/c099-f9p-application-board
And somehow define from the web interface my survey values to initialize one , and when the survey is completed send the correction data with a method I prefer (ntrip, tcp,...)

closed 2024-03-06 by DavideLentini