Stefal/rtkbase #372: RTCM Serial Service

addleman89, 2024-03-09:
I’m sending messages 1005,1008,1077,1087,1230 to the serial service using a RPI. The Pi is connected to an eByte E90-DTU(433L37). My radio is configured to send 240byte packets at 19200bps. My issue is my rover isn’t receiving consistent data so RTK fix comes and goes. Is RTKBase sending more data than my radio can handle?

jaerabr, 2024-03-09:
Hello, I use this configuration on my Lora and it works perfectly with msm4 messages.
(1074.1084.1094.1124)

[image]

addleman89, 2024-03-09:
I wondered if maybe the msm7 messages weren’t too big. I’ll change them over and see if that helps. Thanks!

Stefal, 2024-03-24:
Could we close this issue ?

addleman89, 2024-03-25:
Yes, thank you