Stefal/rtkbase #376: Add TCP client service

Stefal, 2024-03-24:
**Is your feature request related to a problem? Please describe.**
RTKdirect don't use ntrip but tcp client link for the gnss base

**Describe the solution you'd like**
Add tcp client service

**Additional context**
#373

Jef239, 2024-04-10:
I do it for https://rtkdirect.com/. In run_cast.sh 

```
if [[ ${mnt_name_a} == TCP ]] && [[ ${svr_pwd_a} == TCP ]]
  then
    out_caster_A="tcpcli://${svr_addr_a}:${svr_port_a}#rtcm3 -msg ${rtcm_msg_a} -p ${position}"
  else
    out_caster_A="ntrips://:${svr_pwd_a}@${svr_addr_a}:${svr_port_a}/${mnt_name_a}#rtcm3 -msg ${rtcm_msg_a} -p ${position}"
fi

if [[ ${mnt_name_b} == TCP ]] && [[ ${svr_pwd_b} == TCP ]]
  then
    out_caster_B="tcpcli://${svr_addr_b}:${svr_port_b}#rtcm3 -msg ${rtcm_msg_b} -p ${position}"
  else
    out_caster_B="ntrips://:${svr_pwd_b}@${svr_addr_b}:${svr_port_b}/${mnt_name_b}#rtcm3 -msg ${rtcm_msg_b} -p ${position}"
fi

```

Stefal, 2024-04-10:
See https://github.com/Stefal/rtkbase/tree/tcp-client_udp
 A separate service for client TCP connexion will be available in the next release

craisbeck, 2024-04-16:
We looked at your TCP branch but noticed more recently you disabled the UDP services and commented out the setting file in the web app.  We would like to use this feature.  Does this branch work correctly or not?

Stefal, 2024-04-18:
> We looked at your TCP branch but noticed more recently you disabled the UDP services and commented out the setting file in the web app. We would like to use this feature. Does this branch work correctly or not?

TCP Client should work (please inform me if it's not the case).

For UDP, i had some mixed results, and Strsvr seems to have more options than str2str. This is the reason I've disabled it for now.

Stefal, 2024-05-29:
@Jef239 
I've just created an account on rtkdirect, and I've connected my base station with ntrip parameters. I don't understand why you need a tcp client connection.

Did rtkdirect made some changes?

Jef239, 2024-05-29:
> Did rtkdirect made some changes?

YES. RtkDirect is doing a complete infrastructure redesign because the old design can't handle the load. So they switched from TCP to NTRIP v2. But NTRIP v1 is already working. I tested RtkBase with RtkDirect, everything is fine.

Stefal, 2024-05-29:
Ok, so the tcp client service becomes unuseful. I will hide it.