Stefal/rtkbase #381: Add units of measurement to File Service settings

Jef239, 2024-04-10:
This is strange and incomprehensible - there are hours and seconds and days, what is measured in what? It is necessary to enter the units of measurement in the name of the parameters.

Stefal, 2024-04-18:
You can read the units in the tooltips
[image]

Do you think we need to add them in another place?

Jef239, 2024-04-18:
Yes! In the names of parameters!

Neither I nor my tester realized that if you remove a value, you can find out what units it was in. We wanted to see in what units what is set by default.

[image]

Stefal, 2024-06-11:
units, help, added on all services !
example:
[image]

Stefal, 2024-07-05:
fixed in release 2.6.0