Stefal/rtkbase #385: Add U Center Support to Pi image

craisbeck, 2024-04-16:
Is there any way that the Pi image can be set up to forward the USB port info onto the TCP port so a person could use the network function of UCenter to monitor that the station is working correctly?

Stefal, 2024-04-18:
It's already available. Connect U-Center to the base station ip address, on port 5015.

It may change in the future.

closed 2024-04-18 by Stefal

craisbeck, 2024-04-18:
Is that documented somewhere because we never seen it in any readme

Stefal, 2024-04-18:
You can read this issue too : https://github.com/Stefal/rtkbase/issues/188

craisbeck, 2024-04-18:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

We wrote our own node and socat interface quick.   Thats where the comment came from.  But now we will use your build it one.    I have a discourse going because while using your base I am unable to ever get a rtk lock.   I am able to get a rtk lock when using a caster.

I was hoping to be able to use your tcp branch to get local corrections and caster corrections.   I read somewhere that they can alternate back and forth.

The later has nothing to do with not getting a rtk lock.   I’m not sure how to fix that.

Best Regards from my tiny keyboard,
Cory Raisbeck
Boxlogix Automation, Owner