Stefal/rtkbase #396: Web server won't run when using the "update" button to update to latest.

Davidwedel, 2024-05-28:
**To Reproduce**
Steps to reproduce the behavior:
Use the update button to update from 2.4.2 to 2.5. 

**Expected behavior**
On trying to access the webpage service, I get the "Problem loading page" error.

**Software/Hardware (please complete the following information):**
 - RTKBase release number: 2.5.0
 - OS: Ubuntu 22.04.4 LTS aarch64
 - SBC Orange Pi Zero3
 - GNSS Device: Ublox F9P
 - Connection type: usb

**Additional context**
Journalctl logs:
```[my_username]@[my_hostname] ~/r/web_app> sudo journalctl -xeu rtkbase_web
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit rtkbase_web.service has entered the 'failed' state with result 'exit-code'.
May 28 03:06:00 [my_hostname] systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 983.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ Automatic restarting of the unit rtkbase_web.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
May 28 03:06:00 [my_hostname] systemd[1]: Stopped RTKBase Web Server.
░░ Subject: A stop job for unit rtkbase_web.service has finished
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A stop job for unit rtkbase_web.service has finished.
░░ 
░░ The job identifier is 77412 and the job result is done.
May 28 03:06:00 [my_hostname] systemd[1]: Started RTKBase Web Server.
░░ Subject: A start job for unit rtkbase_web.service has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit rtkbase_web.service has finished successfully.
░░ 
░░ The job identifier is 77412.
May 28 03:06:00 [my_hostname] systemd[218463]: rtkbase_web.service: Failed to locate executable /home/[my_username]/rtkbase/venv/bin/python: No such file or directory
░░ Subject: Process /home/[my_username]/rtkbase/venv/bin/python could not be executed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The process /home/[my_username]/rtkbase/venv/bin/python could not be executed and failed.
░░ 
░░ The error number returned by this process is ERRNO.
May 28 03:06:00 [my_hostname] systemd[218463]: rtkbase_web.service: Failed at step EXEC spawning /home/[my_username]/rtkbase/venv/bin/python: No such file or directory
░░ Subject: Process /home/[my_username]/rtkbase/venv/bin/python could not be executed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The process /home/[my_username]/rtkbase/venv/bin/python could not be executed and failed.
░░ 
░░ The error number returned by this process is ERRNO.
May 28 03:06:00 [my_hostname] systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=203/EXEC
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ An ExecStart= process belonging to unit rtkbase_web.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 203.
May 28 03:06:00 [my_hostname] systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit rtkbase_web.service has entered the 'failed' state with result 'exit-code'.
[my_username]@[my_hostname] ~/r/web_app> cd /home/[my_username]/rtkbase/
[my_username]@[my_hostname] ~/rtkbase> ls
archive_and_clean.sh  CHANGELOG.md  check_timesync.sh  data  LICENSE  logs  network_check.sh  README.md  receiver_cfg  rtkbase_update.sh  run_cast.sh  settings.conf  settings.conf.default  tcpdump.log  tools  unit  web_app
[my_username]@[my_hostname] ~/rtkbase> la
total 1.3M
drwxrwxr-x  8 [my_username] [my_username] 4.0K May 17 23:47 .
drwxr-x--- 15 [my_username] [my_username]  20K May 28 03:05 ..
-rwxrwxr-x  1 [my_username] [my_username] 1.4K Dec 28 12:29 archive_and_clean.sh
-rw-rw-r--  1 [my_username] [my_username] 9.3K Jan 30 16:09 CHANGELOG.md
-rwxrwxr-x  1 [my_username] [my_username]  538 Apr 21  2020 check_timesync.sh
drwxr-xr-x  2 [my_username] [my_username] 4.0K Dec 30 04:00 data
-rw-rw-r--  1 [my_username] [my_username]  34K Jun  6  2020 LICENSE
drwxr-xr-x  2 [my_username] [my_username] 4.0K Oct 26  2023 logs
-rwxrwxr-x  1 [my_username] [my_username] 1.5K Apr 21  2020 network_check.sh
-rw-rw-r--  1 [my_username] [my_username]  18K Dec 27 20:05 README.md
drwxrwxr-x  2 [my_username] [my_username] 4.0K Jul  4  2020 receiver_cfg
-rwxrwxr-x  1 [my_username] [my_username]  15K Jan 29 15:13 rtkbase_update.sh
-rwxrwxr-x  1 [my_username] [my_username] 4.2K Dec 28 12:29 run_cast.sh
-rw-rw-r--  1 [my_username] [my_username] 2.1K May 17 23:47 settings.conf
-rw-rw-r--  1 [my_username] [my_username] 4.1K Jan 30 15:58 settings.conf.default
-rw-rw-r--  1 [my_username] [my_username] 1.1M Jan 10 21:52 tcpdump.log
drwxrwxr-x  7 [my_username] [my_username] 4.0K Jan  5 13:25 tools
drwxrwxr-x  2 [my_username] [my_username] 4.0K Dec 27 20:05 unit
drwxrwxr-x 10 [my_username] [my_username] 4.0K May 28 03:00 web_app
[my_username]@[my_hostname] ~/rtkbase> 

```

I'm really not sure why I don't have the venv directory. Does the "update" button not do this, but the install script does?
Ask if you need more info. Thanks!

82Jussi, 2024-05-28:
** To Reproduce**
Steps to reproduce the behavior:
Use the update button to update from 2.4.1 to 2.5.0

**Expected behavior**
On trying to access the webpage service, I get the "Problem loading page" error. The corrections are still working though.

**Software/Hardware (please complete the following information):**
RTKBase release number: 2.5.0
OS: Windows 11 Home 21H2 22000.2538
SBC: Raspberry Pi 3 Model B Plus Rev 1.3
GNSS Device: U-blox_ZED-F9P - 1.13
Connection type: usb

Stefal, 2024-05-28:
Hi! 
> I'm really not sure why I don't have the venv directory. Does the "update" button not do this, but the install script does?
> Ask if you need more info. Thanks!

It should, with line 325:
https://github.com/Stefal/rtkbase/blob/12c49b1a7cce3fd7f8dd194bf4d13a102740bf9d/rtkbase_update.sh#L307-L338

Could you execute this:
`sudo ~/rtkbase/tools/install.sh --dependencies`
then
`sudo ~/rtkbase/tools/install.sh --rtkbase-requirements`

Please write me the errors you will see.

Davidwedel, 2024-05-28:
So I had time to run that this morning, but didn't get the logs all copied. I can send this evening if needed. I had an issue with /var/logs being out of space. That was keeping anything new from installing. I'm not sure why, but sysstat had 40 some MB of data. Even ```sudo apt install ...``` wasn't working. And running the commands you sent definitely installed a bunch of stuff, after id taken care of the storage space issue. I may try installing 2.4 on my laptop, and using the update button to verify that it was the log issue that was causing the problem. 

If that's the case, it was strictly an issue with my system, though a warning that not all was installed might be nice!

Stefal, 2024-06-04:
> If that's the case, it was strictly an issue with my system, though a warning that not all was installed might be nice!

Yeah ! That would be nice, but RTKBase can't manage all the various cases. I'd like to, but I already don't have enough time to what I wanted to add in RTKBase.

So I will say ... PR Welcome 😉

closed 2024-11-23 by Stefal