Stefal/rtkbase #404: Location on status page does not show location data

ttielemans, 2024-06-24:
**Describe the bug**
The longitude, latitude and height are displayed as "Latitude: 0.00000000 ° Longitude: 0.00000000 ° Height: 0.000m" on the status page.

**To Reproduce**
None, present at startup

**Expected behavior**
Long/Lat/Height are displayed

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.5.0
 - OS: [Debian GNU/Linux 12 (bookworm)
 - SBC: Raspberry Pi 5 Model B Rev 1.0
 - GNSS Device: UM980 
 - Connection type: standard UART tttyAMA0 @ 115200 baud

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
RTKBase is serving Onocoy an RTKDirect without issues. GPSMon and CGPS show correct data.

jonasniesner, 2024-07-21:
I habe the same issue with the UM982

geofis, 2024-07-21:
Hello @jonasniesner @ttielemans,

I think that [this issue entry](https://github.com/Stefal/rtkbase/issues/351) could be helpful for you.