Stefal/rtkbase #405: Emlid connection problem

feco9308, 2024-06-24:
Hello! 

The problem is that there are 2 bases that are registered on the caster emlid page.

Regardless, for some reason the connection with the caster emlid is lost. When checked on the web, it also indicates on the emlid page that the base is offline. 

In this case, restarting doesn't help either. Only if the base is off for about 20 minutes. 
Or the other solution is to re-register, let's say, to the "a" server online and then transfer it to the base. 

I think that sometimes the internet is slow and this would cause such an anomaly. 

This happened about a month and a half ago. 

All Raspberry pi 3 and 4 POE power supplies
USB with F9p module. 
Latest rktbase
I am attaching a longer logo that I was able to save.
[Új szöveges dokumentum_log.txt](https://github.com/user-attachments/files/15960363/Uj.szoveges.dokumentum_log.txt)

Thanks for your help!

thebiggestgit, 2024-07-27:
Not experienced this issue with emlid caster myself, running on an RPI4 (currently with RTK Base v2.6.0 after applying manual services fix) with that configured, nor when I was running on a slower Pi 2W.  Maybe a glitch emlid side - is this still happening?

feco9308, 2024-07-30:
There was a short period. He also threw out several bases. Now everything is working again for quite some time. They are with the same internet service provider. 

I suspect a network problem. Because at that time, Caster emlid has high ping values ​​of around 120. 

Also, sometimes the DNS didn't work properly either.

Stefal, 2024-10-06:
It looks like it was not a RTKBase problem. Closing.

closed 2024-10-06 by Stefal