Stefal/rtkbase #409: When trying to access through Home Assistant we get "The CSRF tokens do not match" error

brodhaq, 2024-07-06:
I have added the web interface page as a "Control panel" into my Home Assistant. When I try to open it, I get error "The CSRF tokens do not match".

**To Reproduce**
1. Install Home Assistant (for example on a Raspberry Pi)
2. Go to Settings -> Control Panels
3. Click "Add control panel" and select "web page" type of panel
4. Input your URL of RTKbase web interace (http://192.168.0.194/ in my example)
5. Click the newly created control panel in the menu of the Home Assistant and enter your admin password. The error will appear.

**Expected behavior**
To get standard access to the web interface of RTKBase

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.5.0
 - OS: Home Assistant OS 12.4
 - Raspberry Pi 3B+
 - GNSS Device: not relevant
 - Connection type: Home Assistant RPi is on LAN, RTKBase RPi is on WiFi on the same network.

**Screenshots**
[image]

**Additional context**
Accessing the RTKBase web page through standard web browser works fine

Stefal, 2024-07-09:
Could you try with another browser or in incognito mode?

brodhaq, 2024-07-09:
Tested:
Google Chrome: does not work through Home Assistant
Google Chrome with Incognito mode: does not work through Home Assistant
Firefox: Google Chrome: works perfectly through Home Assistant
iPhone with Home Assistant app: does not work through Home Assistant app

Pavel

DanaMiller-PGH, 2025-09-05:
FYI, I got this to work with this setup:

- Server
-- Raspberry Pi 3 Model B Plus Rev 1.3
-- Debian 12 (Bookworm)
-- U-blox_ZED-F9P - 1.00
- Client (Browsing)
-- MacOS 16.6.1, Safari 18.6
- Home Assistant
-- Running on LCX Host inside ProxMox PVE