Stefal/rtkbase #414: gpsd.service failed to start after update.

Kcoolz, 2024-07-10:
################################
STARTING SERVICES
################################
Job for gpsd.service failed because the control process exited with error code.
See "systemctl status gpsd.service" and "journalctl -xe" for details.

After updating from 1.5 to 2.6 gpsd will no longer start, started fine before updating. killing the proccesses and removing gpsd.socket and ./install.sh --start-services does not help.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.0
 - OS: Linux 11 (bullseye) (RPI 4)
 - SBC RPI 4.
 - GNSS Device: Ublox F9P RTK
 - Connection type: USB

**Screenshots**
[image]
[image]

thebiggestgit, 2024-07-21:
Mine did the same on RPI4.

Stefal, 2024-07-21:
I will try to take some time to look at this bug next week

Stefal, 2024-07-23:
I can't repro with a Raspberry Pi 3, starting with this image: https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/tag/RTKBaseGNSS-RPi-2.4.1 and updating from 2.4.1 to 2.5 and then to 2.6

I will try with a Raspberry Pi 4 later.

Stefal, 2024-07-24:
Test with a Pi 4 and https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/tag/RTKBaseGNSS-RPi-2.4.1
Can't repro

Test with a Pi 4 and https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/tag/RTKBase-GNSS-RPi-2.5.0
Can't repro but found another bug: #418

Kcoolz, 2024-08-29:
I tried again with that image, on a pi4 and 5. Didnt work.
Tried again with a new image and same thing. Still getting the GPSD error, i also tried a different GPS unit.

Stefal, 2024-10-06:
Could you try this: https://github.com/Stefal/rtkbase/issues/431
And write here if you have `127.0.0.1       localhost` in your `/etc/hosts` file?

joaquinveirana, 2025-03-11:
Hi, I get the same error when installing on a fresh Raspberry Pi 4 with the current version of the repo.

Any clue??

sbonaime, 2025-03-11:
> Hi, I get the same error when installing on a fresh Raspberry Pi 4 with the current version of the repo.
> 
> Any clue??

Did you use 2.6.1_1 
https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/tag/RTKBase-GNSS-RPi-2.6.1_1
or RTKBase-GNSS-RPi-2.6.1
[RTKBase-GNSS-RPi-2.6.1](https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/tag/RTKBase-GNSS-RPi-2.6.1)

joaquinveirana, 2025-03-11:
> Did you use 2.6.1_1 https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/tag/RTKBase-GNSS-RPi-2.6.1_1 or RTKBase-GNSS-RPi-2.6.1 [...]

I used [https://github.com/Stefal/rtkbase/releases/tag/v2.6.3](https://github.com/Stefal/rtkbase/releases/tag/v2.6.3)
And 2.5.0 too.
When I run sudo ./install.sh --all release I get the error

Tdorb, 2025-05-05:
Step 1 and the first bit of step 3 here:
https://rtklibexplorer.wordpress.com/2022/11/10/raspberry-pi-based-ppk-and-rtk-solutions-with-rtklib/
After this, I don't have to update.

Aspir1n46, 2025-05-20:
Same problem (

Stefal, 2025-05-22:
It's not a big deal if your base station is connected to internet, but I'll try to repro once again.

> Same problem on 2.6.3 with um980 (
You don't use a F9P, see the first point : https://github.com/Stefal/rtkbase?tab=readme-ov-file#advanced

quada03, 2025-05-27:
my drotek f9p showed up so I plugged that in and there were no issues with the gpsd.service. initially I was using an ardusimple f9p which I can plug back in and recreate the errors repeatedly. so switching to the drotek unit solved the issue for me even though they were both f9p ubx chips

Stefal, 2025-06-02:
@quada03 Thank you, your message helped me to remember that the Ardusimple Usb seems to use a serial/usb adapter, not the F9P usb port.

@Kcoolz On your screenshot inside the first message I see that gpsd didn't start because the RTKBase main service (str2str_tcp.service) was not started. That's an expected behaviour.
Now I need to know if the str2str_tcp service was off for a good reason.