Stefal/rtkbase #418: str2str file not found after 2.5 to 2.6 update (Raspberry Pi 4)

Stefal, 2024-07-24:
**Describe the bug**
After a 2.5 to 2.6 update, the main service doesn't work anymore

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.0
 - OS: Raspberry Pi Os Bookworm aarch64
 - SBC Raspberry Pi 4
 - GNSS Device: F9P
 - Connection: usb

**Additional context**
During the 2.5 to 2.6 update, the script copy the aarch64 files from /tools/bin/rtklib_b34j/aarch64 but these binaries don't run.

workaround: after the update, ssh to the Pi, delete the /tools/bin/rtklib* folders, then launch `sudo ~/rtkbase/tools/install.sh --rtklib`

thebiggestgit, 2024-07-24:
Thanks, tried your workaround above worked for me on RPI4 from the 2.5 image linked on your main page to 2.6. Appears I have the same GNSS receiver too, a U-blox_ZED-F9P connected via USB.

Stefal, 2024-07-24:
Note for me: 
During the next update, test the `str2str -h` exit code, and compil rtklib if the exit code is not 0.

JonathanRubiero, 2024-08-13:
I faced the same issue, and the workaround mentionned worked fine !

commit 94709977 (2024-08-23): fix #418
closed 2024-08-26 by Stefal