Stefal/rtkbase #420: SBAS in satelite grid named without S

Jef239, 2024-07-30:
SBAS in satelite grid named without S. I see it n Mosaic-X5 receiver in RTCM3 mode

[image]

Stefal, 2024-11-23:
I don't know if I'll spend too much time on this graph. A complete rewrite may be a better option as I'd like to add a skyplot.

Jef239, 2024-11-23:
We eventually turned off SBAS. It turned out that Mosaic X5 has few channels and in Europe it is better to turn off SBAS and IRNSS, but receive low satellites of other systems and QZSS.

So it is no longer relevant. But I looked at how it is arranged there - there really is no letter added for SBAS. That is, it was intended that way.