Stefal/rtkbase #423: WITMOTION WTRTK-982 - Main service is off or malfunctioning. Nothing could be displayed here.

enuro12, 2024-08-11:
**Describe the bug**
Main service is off or malfunctioning. Nothing could be displayed here. Is displayed on the web page

**To Reproduce**
Steps to reproduce the behavior:
1. Go to IP of the Raspberry PI

**Expected behavior**
Honestly i have no idea, this is my first build.

**Software/Hardware (please complete the following information):**
 - RTKBase latest
 - OS: Raspberry Pi Os 32bit
 - SBC Raspberry Pi 4
 - GNSS Device: WITMOTION WTRTK-982
 - https://www.amazon.com/dp/B0CY3HYRWG?psc=1&ref=product_details
 - Connection type: USB

**Screenshots**
Here is the screen shot of my GNSS Settings i used.
https://i.imgur.com/sQI882F.png

**Additional context**
Using this reciever with AGOpen appears to function. Don't know how useful that is to know.

geofis, 2024-08-11:
Hello,

Considering that the WITMOTION WTRTK-982 uses the UM982 module from Unicore Communications, and RTKBase was not originally designed for this specific use case, I recommend reviewing the following issues:

Right RAW messages types for RTKbase
https://github.com/Stefal/rtkbase/issues/343#issuecomment-1820650503

Add support for operating Unicore modules (e.g., UM980).
https://github.com/Stefal/rtkbase/issues/351

In addition, I also recommend checking the following discussion from AgOpenGPS:
https://discourse.agopengps.com/t/is-the-um982-an-f9p-killer-for-aog/13960

Normally, I use RTKBase with several UM980 modules (the 'sibling' of the UM982, with a single antenna) to set up temporary bases, and the results are good. In one of the issues I mentioned above (the one I initiated), I discuss the branch I created in my fork of RTKBase to handle the UM980. However, the user @Jef239 developed the necessary features to integrate Unicore receivers into RTKBase (also mentioned in the referenced issue).

I’m not sure what application you are pursuing with the 982, but if it’s not related to an auto steering system, I wouldn’t recommend using a Raspberry Pi 4, as they consume a lot of power.

Jef239, 2024-08-12:
For UM982 you can use [GNSSOEM/UM980_RPI_Hat_RtkBase](https://github.com/GNSSOEM/UM980_RPI_Hat_RtkBase) . There will be a new release based on the new version of RtkBase in the coming days. You cam see it in dev branch.

enuro12, 2024-08-12:
@geofis Thank you, I'm reading the other information you provided. Would a Pi 3 or 5 do better? 

@Jef239 Thank you!