Stefal/rtkbase #425: How to configure Quectel LC29HBS

jupiter9595, 2024-08-26:
Waveshare has refered RTKBase in their LC29HBS board description https://www.waveshare.com/wiki/LC29H(XX)_GPS/RTK_HAT

Looks like their configuration was successful but they did not show  Main Service configuration details.

Has anyone successfully configure this GNSS unit?

RTKBase does not detect this unit.
How to correctly configure this unit?
Are Base coordinates fill in automatically when unit is detected?
I put the following coordinates: 49.208601 19.072489 1050    and   RTKBase says format is not correct.

Please help to configure RTKBase unit

jupiter9595, 2024-08-26:
Coordinates format I've figured out myself by looking at input fields defined pattern. Altitude should have at least 2 decimal places 1050.00 in this case

deece, 2024-09-07:
LC29H(BS) Initial Configuration:
Connect to the unit using QGNSS. In the console, data input, Advanced, select: Text format: ASCII, Checksum Type: NMEA, Suffix: CRLF

In Data Input, enter the following  (as per Quectel_LC29HBS_GNSS_Protocol_Specification_V1.1-1.pdf):

Check firmware version:
`$PQTMVERNO*`

Enable MSM7 messages
`$PAIR432,1*`

Enable Station Reference Message 1005
`$PAIR434,1*`

Enable Ephemeris messages
`$PAIR436,1*`

Crank the baud rate up (Could also go 3000000)
`$PAIR864,0,0,921600*`

Enable NMEA GGA Time, position, and fix related data
`$PAIR062,0,1*`

Enable NMEA GLL Position data: position fix, time of position fix, and status
`$PAIR062,1,1*`

Enable NMEA GSA GPS DOP and active satellites
`$PAIR062,2,1*`

Enable NMEA GSV Satellite information
`$PAIR062,3,1*`

Enable NMEA RMC Position, velocity, and time
`$PAIR062,4,1*`

Enable NMEA VTG Track made good and speed over ground
`$PAIR062,5,1*`

Enable NMEA ZDA UTC day, month, and year, and local time zone offset
`$PAIR062,6,1*`

Enable NMEA GRS GRS range residuals
`$PAIR062,7,1*`

Enable NMEA GST Position error statistics
`$PAIR062,8,1*`

Save parameters
`$PQTMSAVEPAR*`

Power cycle the unit, reconnect at the new baud rate

Start Survey-in (1 day duration)
`$PQTMCFGSVIN,W,1,86400,0,0,0,0*`

You will see PQTMSVINSTATUS repeating, when the third field=2, the survey-in is complete. Fields 8, 9 & 10 are the mean ECEF coordinates. Convert these to lat/long/elevation above ellipsoid to enter into RTKBase. You may have to prune some places after the decimal point for height before the parser will accept it:
```
import numpy as np

# Constants
a = 6378137.0  # WGS-84 Earth semimajor axis (m)
f = 1 / 298.257223563  # WGS-84 flattening
e2 = f * (2 - f)  # Square of eccentricity

# Input ECEF coordinates (X, Y, Z)
X, Y, Z = -123456789.0000, 1263456789.000, -123456789.1234

# Calculate longitude
longitude = np.arctan2(Y, X)

# Iterative calculation of latitude and height
p = np.sqrt(X**2 + Y**2)
lat = np.arctan2(Z, p * (1 - e2))  # Initial guess for latitude
lat_prev = 0
h = 0

# Iterate until convergence
while abs(lat - lat_prev) > 1e-12:
    lat_prev = lat
    N = a / np.sqrt(1 - e2 * np.sin(lat)**2)  # Prime vertical radius of curvature
    h = p / np.cos(lat) - N
    lat = np.arctan2(Z, p * (1 - e2 * N / (N + h)))

# Convert latitude and longitude to degrees
latitude = np.degrees(lat)
longitude = np.degrees(longitude)

latitude, longitude, h
```

When complete, save parameters
`$PQTMSAVEPAR*`

deece, 2024-09-07:
For detection:
`$PQTMVERNO*`

returns:
`$PQTMVERNO,LC29HBSNR11A01S,2023/02/13,10:14:06*`

The pertinent information to detect this is:
`LC29HBS`

jupiter9595, 2024-09-07:
@deece - thanks a lot for this comprehensive guide! Have you ever got RTK fix status with this board? I read in internet, people get RTK float only.

deece, 2024-09-07:
I only got my base station functional tonight, I probably won't be at a point where I have rover for months.

deece, 2024-09-19:
@jupiter9595 Care to give this PR a try? https://github.com/Stefal/rtkbase/pull/432

You can find the full tree here: https://github.com/deece/rtkbase/tree/lc29hbs

benpeterson40, 2024-10-30:
What would be command to install the lc29hbs branch. 

I did get the lc29hbs and LG290P working with rtkbase  but was interested in checking out your more automated install process.

Waqasop, 2025-02-10:
@jupiter9595  hey were you able to get rtk fix with this module as i am kinda stuck at float

RaptorSDS, 2025-06-02:
Please add this Module to rtkbase , or comit your tree to oginal master

RaptorSDS, 2025-06-04:
> Just wanted to add to this great write up by [@deece](https://github.com/deece), whether it's on Windows or Linux (raspberry pi). I've had to have the LC29H-BS connected via USB to issue commands and receive a response I can actually read. Whereas if I manually send the commands over I2C on the Pi, responses come back delayed and garbled. But to be clear, if its running on I2C the proposed pull request [#432](https://github.com/Stefal/rtkbase/pull/432) seems to handle this behind the scenes.
> 
> [@RaptorSDS](https://github.com/RaptorSDS) - if i had to wager a guess (but I could be wrong as a lot of this is new to me as well), it sounds like you need to manually register the configure_gps.service to get the device to start outputting data as the service sends the commands above on boot since they are not saved to the device's flash memory.

Thank you after edit settings.conf and use configure_gps.services (chmod +x) now it works 
be aware in setting you have to set Quectel LC29HBS as name , in Webinterface this name ist not possible red field

MinimumDowntime, 2025-06-04:
Thanks for confirming my steps @RaptorSDS 😃

ezequielchan, 2025-08-10:
Hey @deece, really appreciate your work — totally deserves a merge 😄. Finally not stuck with just Null Island coords!

erok26, 2025-11-10:
@deece great work on updating the LCH29BS. I'm able to use the base and get into RTK Fixed state on my rover, however, I'm seeing a behavior where when in DGPS and RTK Float, it is off by .00015 in latitude and .00001 in longitude. I've rebuilt the base 6 or so times and have changed coordinates a few times in the base to try and adjust for the off coordinates. I cannot see any reason that each time I rebuild it, I am still off by some 10 to 40 meters. I'm also struggling with similar issues as others where in the status page, I see the satellites signal quality graph, but in the bottom the base coordinates are not being detected. Could this be the reason? Is anyone able to list out the steps required to properly configure the base using the LCH29BS? I'm trying to survey my property lines using a known coordinate for the base, but I'm stuck being that when in RTK Fixed its is pushing my rover off by several meters.

RaptorSDS, 2025-11-10:
I done the survey in by disconnect the base , connect my rover to the fixed base antenne used another base in my near to have trk fix for this position wait 10min to see the avarage and use the now found position for base antenna .
after that switching back to base chip+antenna and iput this in the base coordinate .
after many other problems with wlan driver in rasberry debian linux i now switch to another box with openwrt and str2str

michel2711, 2025-11-15:
Answering my own question. I found some of the files that I missed during the install from deece's tree. Moving ahead :)

michel2711, 2025-11-16:
I used the lc29hbs branch in Deece. It seems to contain most of the files that are necessary to configure the receiver. So far, I'm still not detecting the receiver and I'm wondering if the receiver type (Quectel LC29HBS) is correct. The web UI seems to disagree. 
However, all services are running and the web application starts. That's what I did:

1 - I got a 64bits bookworm image with desktop using PI imager
2 - apt-get update and upgrade
3 - raspi-config to enable I2C and UART hardware
3 - cd ~
     wget https://raw.githubusercontent.com/deece/rtkbase/refs/heads/lc29hbs/tools/install.sh -O install.sh
     chmod +x install.sh
4 - I ran the manual install (steps 1 to 10)

Note that for some reason that I cannot explain, I had to use a virtual environment on step 4 ( I also know close to nothing about Python).
Also, the step 10 seems to expect configure_gps.sh to be in ~/rtkbase and it's in the ~/rtkbase/web_app directory. It needs to be moved up and chmod'ed +x etc. After that step 10 works fine and the services are all running. 
I am now on the web app UI and it cannot detect the receiver, although I see the receiving led blinking during the detect phase . I'm going to look at the code and try to understand what's happening. Hopefully, more news in the next few days.

deece, 2025-11-18:
Sorry, I haven't built a rover yet

erok26, 2025-11-21:
@ckemere,
I have not been able to get the autodetect on the web ui to work at all. That is built by stefal and I don't think supports the LC29HBS chips. In addition, if you use the configure_gps.sh, you need to make sure that you edit the settings.conf and set the Receiver type: to Quectel LC29HBS, the web UI will not agree with it due to the space in it, but the configure_gps.sh should then work. I went ahead and updated the install.sh and the configure_gps.sh to support a name with an _ in it so that the web ui doesn't complain about it. However, it wasn't until I rewrote the survey-in python script to run without pyproj and issue all of the serial command to the device to turn on all the nmea options that I realized it was now polling GPS locations on the map in the web ui status page.

ckemere, 2025-11-21:
One more edit - it works much better if this line has a shorter timeout:
```
        with serial.Serial(port, baudrate=speed, timeout=timeout) as ser:
```
to
```
        with serial.Serial(port, baudrate=speed, timeout=0.1) as ser:
```
With this change, I can run the full install script from @deece and get something running without errors. I have yet to test it with RTK!