Stefal/rtkbase #427: Not working at all in Raspberry Pi 1 model B+ ?

flyingsurveyor, 2024-08-28:
Hi!
I have got an old raspberry model B+
I've tried several times to install Rtkbase on it.
Tried with a fresh image of Raspberry pi os (Bookworm, complete and lite versions). Tried with the iso image that you recommend here.
Nope. 
Services seem not loading or starting, they seem not exist at all ("rtkbase_web.service not found").
I tried to change ports on my router, I tried the manual installation. Nope....
The strange fact is that I already use rtklib on this old raspberry board, in particular str2str or for ubx logging with python codes....
My two boards ardusimple zed-f9p (I have two of them) are recognized and working properly... In fact they were recognized during the rtkbase installation phase....
For now the only result obtained are the parameters of my boards completely ravaged and changed......
Not a problem anyway... Luckily I've backup configuration files...

Could it be that this old Raspberry model is absolutely not compatible?

Thanks and sorry for wasting your time replying to me.

DavideLentini, 2024-10-02:
Hey , i don't know if is still interesting to you , but i'ts an hell also for me , installing all the stuff in my Rpi 2 Model B. I came to a conclusion with a sh script that maybe (maybe...) can install all the thing . But my advice is , just buy an RPi5

DavideLentini, 2024-10-02:
And after a 3 hour , this script made all the thing work : 

```
sudo apt update
sudo apt install python3-pip
sudo apt upgrade
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
export CARGO_HOME=$HOME/.cargo
export PATH=$CARGO_HOME/bin:$PATH
pip install cryptography
sudo apt-get install pkg-config
sudo apt-get install libjpeg-dev zlib1g-dev
python3 -m pip install Pillow
pip  install --upgrade pip setuptools 
sudo apt-get install build-essential libssl-dev libffi-dev python3-dev cargo pkg-config
#INSTALL RTKBASE 
wget https://raw.githubusercontent.com/Stefal/rtkbase/master/tools/install.sh -O install.sh
chmod +x install.sh
sudo ./install.sh --all release
```

Stefal, 2025-01-28:
@flyingsurveyor 
Yeah, some python modules are hard to install on old board.

@DavideLentini Thanks for the script.

closed 2025-01-28 by Stefal