Stefal/rtkbase #431: GPSD fails to connect to STR2STR

deece, 2024-09-07:
**Describe the bug**
GPSD fails to connect:
```
Sep 07 20:15:03 rtkgps gpsd[1290]: gpsd:WARN: PACKET: packet_get1(fd 8) errno: Connection refused(111)
Sep 07 20:15:03 rtkgps gpsd[1290]: gpsd:WARN: CORE: device read of tcp://localhost:5015 returned error or packet sniffer failed sync (flags {ERROR})
```

**To Reproduce**
Steps to reproduce the behavior:
Enable '-D 2' in the GPSD_OPTIONS of `/etc/default/gpsd`
Restart gpsd
journalctl -f -u gpsd

**Expected behavior**
GPSD connects to the cloned GPS unit

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.1
 - OS: Raspberry Pi OS Lite, Debian GNU/Linux 12 (bookworm)
 - SBC Raspberry Pi 4
 - GNSS Device: Quectel LC29HBS
 - Connection type: serial

**Additional context**
Replacing 'localhost' in /etc/defaults/gpsd with '127.0.0.1' fixes the problem.

Local resolution appears to be working:
```
root@rtkgps:~# nslookup localhost
Server:		10.0.1.1
Address:	10.0.1.1#53

Name:	localhost.lan
Address: 127.0.0.1
Name:	localhost.lan
Address: ::1
```

It could be because str2str is only bound to the ipv4 address, and not ipv6 too:
```
root@rtkgps:~# netstat -lp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
<snip>
tcp        0      0 0.0.0.0:5015            0.0.0.0:*               LISTEN      676/str2str
<snip>
```

Stefal, 2024-10-05:
Hi!

I don't understand why it doesn't work with localhost.
Is `127.0.0.1       localhost` in your `/etc/hosts` file?

deece, 2024-10-15:
@Stefal newer distros resolve to both IPV6 and IPV4:
```
deece@rtkgps:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

deece@rtkgps:~ $ host localhost
localhost.lan has address 127.0.0.1
localhost.lan has IPv6 address ::1

deece@rtkgps:~ $ cat /etc/hosts
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.1.1	rtkgps
```

shaneshort, 2025-12-09:
I just hit the same thing-- I had to change the GPSD option to use 127.0.0.1 instead of localhost because it doesn't str2str doesn't listen on ipv6.