Stefal/rtkbase #438: Any way to add new Quectel LG290P module?

splee777, 2024-10-09:
I currently run RTKBase with the F9P and it's a really nice system. I recently acquired a Quectel LG290P module to play with and was trying to get it to work with RTKBase. I can get it to send the RTCM3 packets to the the program and get the Main service to show green, but when I go to the status screen, it says not active. And when I try to activate a caster, it comes up orange. 

I had set the main screen protocol to RTCM3, and selected the appropriate tty, but am not sure what else I have to do or if it will even work as is. 

Hopefully someone more capable than me is interested enough to take a shot at this...

benpeterson40, 2024-10-10:
I have a lc29h-bs and LG290P on order. Hopefully the lg290p will only require a few changes from deece's lc29h-bs rtkbase code.

splee777, 2024-10-11:
The commands are different and also the LG290P has a Rover and Base station mode. When it is in Base station mode, it only sends RTCM messages and when you try to enable GGA, GLL etc NMEA messages, it sends them but there is no position data in them. I also tried it in Rover mode and then tried to enable the RTCM messages like on the LC29h. It will only output 1005, 1006 and 1019 but I cannot seem to enable any MSM messages. Oh well, hopefully someone more knowledgeable than me can figure it out....

jaerabr, 2024-10-11:
Hello, I'm using LG290P in rtkbase, I configured it as follows
modo base
$PQTMCFGRCVRMODE,W,2*29

activate the Ephemeris
$PQTMCFGMSGRATE,W,RTCM3-1019,1*54
$PQTMCFGMSGRATE,W,RTCM3-1020,1*5E
$PQTMCFGMSGRATE,W,RTCM3-1042,1*5A
$PQTMCFGMSGRATE,W,RTCM3-1046,1*5E

activated nmea messages
$PQTMCFGMSGRATE,W,GGA,1*0A
$PQTMCFGMSGRATE,W,RMC,1*17
$PQTMCFGMSGRATE,W,GLL,1*0C
$PQTMCFGMSGRATE,W,VTG,1*0E
$PQTMCFGMSGRATE,W,GSA,1*1E

disable rtcm 1005, so that rtkbase can send the corrected coordinate
$PQTMCFGMSGRATE,W,RTCM3-1005,0*59

I use it as a base and rtk uhf rover in the field without internet connection

[image]

benpeterson40, 2024-10-11:
Maybe firmware between your two is different versions?

jaerabr, 2024-10-11:
to see the version
$PQTMVERNO*58

mine is on the latest version
$PQTMVERNO,LC29HAANR1103S,2023/12/14,10:12:35 34

I have the version update file, but I haven't updated it. I already purchased my updated version, but the files were sent to me by george.gao
from Quectel support.

splee777, 2024-10-11:
I got it to work by using just your commands. I'm not sure why it didn't work when I did pretty much the same steps. Maybe I enabled some extra stuff or something. But anyway, thanks!

I have the latest fw version too, but your version above seems to be for a LC29H module?

[image]

jaerabr, 2024-10-11:
> I have the latest fw version too, but your version above seems to be for a LC29H module?
My mistake, I copied it from the email for illustrative purposes only  :)

splee777, 2024-10-11:
I did use that command, and apparently, QGNSS is telling me it is still receiving MSM 4 messages, but when I look at my NTRIP client, it shows it is receiving MSM 7 messages! Something wrong with the Quectel program?

On another note, although RTKBase appears to be working, I'm having difficulty achieving fix. It will sometimes go into float but never achieve fix. Did you have any receiver options, like -TADJ?

jaerabr, 2024-10-11:
> Eu usei esse comando e, aparentemente, o QGNSS está me dizendo que ainda está recebendo mensagens MSM 4, mas quando olho para meu cliente NTRIP, ele mostra que está recebendo mensagens MSM 7! Algo errado com o programa Quectel?
> 
> Em outra nota, embora o RTKBase pareça estar funcionando, estou tendo dificuldade em obter a correção. Às vezes, ele entra em float, mas nunca obtém a correção. Você tinha alguma opção de receptor, como -TADJ?

it only activates after restarting the module, I leave the receiver option blank

splee777, 2024-10-11:
What do you have in the Local Caster RTCM Message section? It appears that If I put 1074 there, it appears as 1074 in the client, and if I put 1077, it shows up as 1077, regardless of what protocol the LG290P sends.

In QGNSS, not matter what I do, save, restart, etc, Qconsole never shows any MSM 7 packets in Messages statistics...

closed 2024-10-13 by splee777