Stefal/rtkbase #441: One Apply button at "GNSS Receiver detection/configuration", but you can't click on it.

Jef239, 2024-10-25:
**Describe the bug**
One Apply button at "GNSS Receiver detection/configuration", but you can't click on it.

**To Reproduce**
Steps to reproduce the behavior:
1. Disconnect the receiver from the Raspberry Pi.
2. Start configuration.
3. Start detection.
4. See only one disabled button

**Expected behavior**
A clear and concise description of what you expected to happen.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.2

**Screenshots**

[image]

**Additional context**
detectCancelBtnElt.remove(); at  $('#configure_receiver_button').on("click", function () in settings.js 

d

Jef239, 2024-10-25:
[commit-06225e7.txt](https://github.com/user-attachments/files/17527074/commit-06225e7.txt)

Patch for fix

Stefal, 2024-11-24:
thanks for the bug report and the patch

commit 18f32e9a (2024-11-24): fix #441
closed 2024-11-24 by Stefal