Stefal/rtkbase #442: RtkBase Version 2.6.2

GianfrancoGaggiano, 2024-10-29:
The whole team did a great job!
 I should point out that sometimes, for some reason, the main service no longer works.
 It seems to me to occur when you temporarily lose your internet connection.

Stefal, 2024-11-20:
Hi!
The internet connection should have no effect on the main service. It's usually a gnss receiver to sbc cable problem. Please check it.

GianfrancoGaggiano, 2024-11-22:
Hi thank you for reply, can you add a function that automatically try to reactivate main service in case it goes off?

Stefal, 2024-12-22:
It works already like that:
`Restart=on-failure`
https://github.com/Stefal/rtkbase/blob/d799c5e75b4d917c985a0e50d6627cf824c5c88c/unit/str2str_tcp.service#L11C1-L11C19

closed 2024-12-22 by Stefal