Stefal/rtkbase #452: Detect and configure GNSS fails

jpucheu, 2025-01-23:
**Describe the bug**
The configuration of the Zed_F9P gnss fails. Doing manually, it reveals that configure doesn't work because there is a problem with ubxtool. 
If I run ubxtool -v, the system respond: ubxtool: failed to import gps, check PYTHONPATH

**To Reproduce**

sudo ./install.sh --detect-gnss --configure-gnss
user for RTKBase is:  juan
################################
USB GNSS RECEIVER DETECTION
################################
/dev/ttyGNSS  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver -  115200
GNSS RECEIVER DETECTED: /dev/ttyGNSS  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver  -  115200
################################
CONFIGURE GNSS RECEIVER
################################
ubxtool: failed to import pyserial
Traceback (most recent call last):
  File "/home/juan/rtkbase/tools/sept_tool.py", line 7, in <module>
    from septentrio.septentrio_gnss.septentrio_cmd import *
  File "/home/juan/rtkbase/tools/septentrio/septentrio_gnss/septentrio_cmd.py", line 2, in <module>
    from . serial_comm import SerialComm
  File "/home/juan/rtkbase/tools/septentrio/septentrio_gnss/serial_comm.py", line 2, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
No Gnss receiver has been set. We can't configure
juan@raspberrypi:~ $ ubxtool --help
ubxtool: failed to import gps, check PYTHONPATH

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.2
 - OS: Raspberry pi OS Lite 64 bit (Debian Bookworm)
 - SBC Raspberry pi Zero 2W
 - GNSS Device:  Ublox F9P
 - Connection type: usb

Stefal, 2025-01-25:
Hmm, that's strange.
If you enter `python3 -m serial`, do you get a list of the serial port?

Uome, 2025-03-11:
I ran into the same problem. My Ublox F9P module functions normally in U-center on windows. When connected to my Pi 4 (running jancelin's image), `lsusb` finds it at _ttyUSB0_. However, when I set _ttyUSB0_ as the Com port and try to connect and/or configure, I receive: "No GNSS receiver detected".

When I enter `python3 -m serial`, I get:

_ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (wrong ELF class: ELFCLASS32): ignored.

--- Available ports:
---  1: /dev/ttyUSB0         'FT232R USB UART - FT232R USB UART'
--- Enter port index or full name:_

Uome, 2025-03-15:
In order to see whether the problem was related to the aforementioned image, I installed the latest image of PiOS and followed the instructions for Easy Installation, but to no avail. When trying to detect my module, I keep receiving: 

> No GNSS receiver detected

The same is true when I try to detect another one of my F9P modules, it also shows up using `ls /dev/tty*` yet isn't found using the detect and configure button.

Stefal, 2025-03-16:
As soon as I have a F9P in hand, I'll test it with a Raspberry Pi.

karcsesz, 2025-04-01:
Running into the same issue with a Raspberry Pi 3 B and following the guide from https://docs.centipede.fr/docs/base/Installation.html

Ended up patching it by
1. Manually apt installing python3-gps
2. Editing the GNSS module detection loop in the install script so it doesn't just search in serial0 and a few ttyS devices but also ttyUSB0 (the F9P was connected via an FTDI converter)

Stefal, 2025-04-01:
But... ttyUSB0 is already in the list : https://github.com/Stefal/rtkbase/blob/03f614afc7e2afd9fa40f155e6d1b568940ea9c9/tools/install.sh#L428C9-L428C88

karcsesz, 2025-04-01:
It's entirely possible the guide I mentioned links to an older build or an image that's been altered somehow. Pardon the noise.
Though hopefully this is something that the image @Uome are running has in common with mine? A simple packaging/dependency issue

jpucheu, 2025-04-01:
I installed manually the package and resolved a dependency issue.
Unfortunately, I didn't document the process, and I don't feel secure to
guide it. I will back to this later

Uome, 2025-04-15:
@karcsesz I'm not sure which processes python3-gps is supposed to affect, but after a `sudo apt-get update` and a `sudo apt-get install python3-gps`, my Pi 4 no longer boots. The green LED stays on for minutes after boot, after which only the red LED remains lit, the device no longer shows anything on its HDMI output, and is no longer reachable through SSH.

Stefal, 2025-04-20:
I did some tests, and I didn't reproduce the problem:

SBC : Rapsberry Pi 4B
Os : Raspberry Pi os lite 64 bits
- F9P connected with its Usb port : ✅
- F9P connected on Pi uart pins : ✅ 
Os : jancelin image_2024-10-04-RTKBase-2-6-1_1_bookworm_64-lite
- F9P connected with its Usb port : ✅ 
- F9P connected on Pi uart pins : ✅ (see below)

For the image_2024-10-04-RTKBase-2-6-1_1_bookworm_64-lite image, I had to : 
 - `sudo apt-get remove raspi-copies-and-fills` to remove the ELF32CLASS errors
 - Reboot
 - enable the uart with `sudo raspi-config`
 - Reboot
 - Then launch `sudo ./install.sh --detect-gnss --config-gnss`

Stefal, 2025-04-20:
> [@karcsesz](https://github.com/karcsesz) I'm not sure which processes python3-gps is supposed to affect, but after a `sudo apt-get update` and a `sudo apt-get install python3-gps`, my Pi 4 no longer [...]

Your problem is not related with the python3-gps installation.