Stefal/rtkbase #457: Error running NRcan rinex observation

AtnDPO, 2025-02-12:
When running a NRCAN rinex file out of X5 receiving ERROR mail. Error : A duplicate data block has been found in your RINEX observation file at epoch 2025 02 11 00 41 30.00. Please correct and resubmit.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.2
 - SBC Orange pi zero 3
 - GNSS Device: simpleRTK3B Pro antenna beitian bt-800S
 - Connection type: USB

Stefal, 2025-02-12:
Hi!
Which Rinex template did you use?

Stefal, 2025-02-16:
I can repro this problem. It's related to something wrong in rtklib b34j.
If you want, you can test this workaround : 
```
cd ~
rm install.sh
wget https://raw.githubusercontent.com/Stefal/rtkbase/refs/heads/rtklib_b34k/tools/install.sh -O install.sh
chmod +x install.sh
sudo ./install.sh --rtklib
```

It will stop the various services, and upgrade rtklib to release b34k.
Then, create the rinex file again.

Please keep me informed.

AtnDPO, 2025-02-16:
Hi Stefan,Do I need to download it on the orangpi? Can I do it over POE? Sorry not the most knowledge. Is it coming in an update? If got 6 stations like this one so I’m looking for the best solution. Déjan

AtnDPO, 2025-02-20:
Still no suc6 for me... i can not even create rinex in RTKBASE

AtnDPO, 2025-03-01:
The solution works! Thanks 😁

closed 2025-03-01 by AtnDPO

Stefal, 2025-03-02:
I reopen this ticket to not forget that I should update RTKLib for the next release.

Stefal, 2025-12-04:
RTKBase 2.7.0 with RTKLib 2.5 is available