Stefal/rtkbase #461: add u-blox ZED-X20P module support in rtkbase

yannick56, 2025-03-09:
English :

Can you add support for u-blox ZED-X20P module in rtkbase ?
This module has triple frequency band and Galileo HAS support. 

simpleRTK4 Optimum with ZED-X20P module
EN : https://www.ardusimple.com/product/simplertk4-optimum-zed-x20p/

How to configure u-blox ZED-X20P
https://www.ardusimple.com/how-to-configure-u-blox-zed-x20p/

Comparison of u-blox ZED-F9P and ZED-X20P
https://www.ardusimple.com/comparison-of-u-blox-zed-f9p-and-x20p/

New product: simpleRTK4 Optimum with u-blox ZED-X20P
https://www.ardusimple.com/new-product-simplertk4-optimum-with-u-blox-zed-x20p/

ZED-X20P module
https://www.u-blox.com/en/product/zed-x20p-module
u-blox X20 all-band high precision GNSS module

ZED-F9P and ZED-X20P - comparison
https://www.u-blox.com/en/zed-x20p#index-05

ZED-X20P series
https://content.u-blox.com/sites/default/files/documents/ZED-X20P_ProductSummary_UBXDOC-304424225-18238.pdf

u-blox GNSS product overview
https://content.u-blox.com/sites/default/files/products/documents/GNSS-product_Overview_UBX-14000426.pdf

All-band vs. Quad-band vs. Triple-band: What’s the real difference?
https://www.ardusimple.com/allband-vs-quadband-vs-tripleband-the-real-difference/

-----------

Français / French :

Pouvez-vous ajouter la prise en charge du module u-blox ZED-X20P dans rtkbase ?
Ce module peut recevoir sur 3 bandes de fréquence et prend en charge Galileo HAS.

simpleRTK4 Optimum avec ZED-X20P module
FR : https://fr.ardusimple.com/product/simplertk4-optimum-zed-x20p/

Comment configurer u-blox ZED-X20P
https://fr.ardusimple.com/how-to-configure-u-blox-zed-x20p/

Comparaison de u-blox ZED-F9P et ZED-X20P
https://fr.ardusimple.com/comparison-of-u-blox-zed-f9p-and-x20p/

Nouveau produit : simpleRTK4 Optimum avec u-blox ZED-X20P
https://fr.ardusimple.com/new-product-simplertk4-optimum-with-u-blox-zed-x20p/

Merci pour ton travail de développeur de rtkbase et ta conférence au Séminaire AgroTIC à 55mn à https://www.youtube.com/watch?v=dUJ46bKSlSE

Stefal, 2025-03-09:
Sure, I will add it when I could get one.

yannick56, 2025-11-14:
@Stefal 
@jancelin 

Can you update with the latest patch (August 16) :
1) the source code at https://github.com/CentipedeRTK/rtkbase 
AND
2) the file "Raspberry Pi 64b: [BaseGNSS-RPi-2.6.1_1](https://github.com/CentipedeRTK/pi-gen_RTKbase/releases/download/RTKBase-GNSS-RPi-2.6.1_1/image_2024-10-04-RTKBase-2-6-1_1_bookworm_64-lite.zip)  in the webpage https://docs.centipede.fr/docs/base/Installation with this patch to add u-blox ZED-X20P module support in rtkbase ?