Stefal/rtkbase #463: Bug: str2str_tcp service fails silently due to background process execution

GeorgeTownsendd, 2025-03-21:
## Observed Behavior
When the service fails to connect to GPS hardware, logs show:
```
Mar 21 04:17:05 gnss-host2 run_cast.sh[451]: stream server start error
Mar 21 04:17:05 gnss-host2 systemd[1]: str2str_tcp.service: Deactivated successfully.
Mar 21 04:17:05 gnss-host2 systemd[1]: Started str2str_tcp.service - RTKBase Tcp.
```

Despite the connection error, systemd reports the process exited with success and doesn't trigger the `Restart=on-failure` directive.

## Technical Cause
The script launches str2str as a background process:
```bash
${cast} -in ${!1} -out ${out_tcp} -b 1 -t ${level} -fl ${logdir}/str2str_tcp.log &
```

This causes the parent script to exit with success (0) regardless of whether the background process connects successfully. Systemd interprets this exit code as a successful service completion rather than a failure.

## Impact
- Dependent gpsd service fails to start: `ExecStartPre=systemctl is-active str2str_tcp.service (code=exited, status=3)`
- GPS functionality unavailable until manual intervention
- Automatic recovery mechanisms ineffective despite being configured

## Workaround
For now I have set ```Restart=Always``` in the systemd file. This works because the only time the str2str exits is when there is a failure. I think it relies on how systemd handles ```Type=forking``` (it watches the subprocess after the parent exits). 

## Thanks
Thanks very much for this project- we are using it at the University of Newcastle to help study GNSS interferometric reflectometry.

Stefal, 2025-04-20:
Thank you for the report.

I remember I've tested many configurations with `Type=forking/simple/...` and launching str2str in background or not. But it's old, and I don't remember why I ended up with forking and running str2str in a background process.

After a few quick tests, it seems to work correctly with `Type=simple` and removing the `&` at the various str2str line end in `run_cast.sh`

I'd be happy if you could test this way and give me some feedback.

fadilom, 2025-04-21:
I am running into the same error and I am not able to follow your instructions because I am so deep into programming. 
I use Raspberry Pi 3 Model B Rev 1.2 with Debian Bookworm. 

[image]

What shall i do exactly to get it running ?

Stefal, 2025-04-21:
@fadilom It's not related to the reported bug. Please create another issue if you need.

PR #498 (2025-09-16): change service type from forking to simple
If the receiver serial port isn't available when the service str2str_tcp starts, it could failed, stop, and never restart.
With this update, the service stop with a failure exit, and then it will try to restart.

it should fix #463 and maybe #483