Stefal/rtkbase #464: Feature Request: Add GPS Survey-In Mode with Time & Accuracy Settings

dremeier, 2025-03-21:
Hi Stefal,

First of all, thank you for creating and maintaining rtkbase — it's a really great project and runs very well on my Raspberry Pi!

I would like to suggest a new feature that I believe would be very useful for many users: a **GPS Survey-In mode** to help determine accurate base station coordinates.

### 🧭 Feature Description

The idea is to add a section to the **status page** that allows the user to perform a Survey-In. The user could specify:
- A **duration** (in seconds) for the survey run
- A **target accuracy** (in meters)

While the survey is running, the current estimated position and precision would be shown. Once the desired accuracy is reached or the time has elapsed, the user would be able to click a **"Use this position"** button that copies the averaged coordinates into the settings page (and ideally saves them into `settings.conf`).

### 🛠️ Example Use Case
This would make it very easy to deploy a base station in the field without having to manually average NMEA output or use RTKLIB directly. A survey mode like this exists in u-center and other tools, but having it directly integrated into rtkbase would simplify workflows greatly.

### 🙏 Why it's useful
- No need for external tools or manual position averaging
- Consistent and reproducible position setup
- Improved ease-of-use for less technical users

If you're open to the idea, I'd be happy to help test it or even contribute code for it.

Thanks again for your work!

Best regards,  
Andre

tranhien1612, 2025-04-23:
@dremeier I want to do same your idea, have you done it?

windkh, 2025-04-23:
Good idea. It would be really accurate if you could define another ntrip server as reference station. The base could temporarily switch to client mode and simply measure the location in survey mode.

dremeier, 2025-04-23:
> [@dremeier](https://github.com/dremeier) I want to do same your idea, have you done it?

Yesterday I gave up, too complicated for me :-(

Stefal, 2025-05-22:
If I've never added this feature, it's partly because it won't give you coordinates in your local reference system (ETRF2000 for europe) but in ITRF@now, you will have an offset (more than 80cm in europe, in 2025). And It means that if you survey the same point in a few years, the coordinates won't be the same.

Do you still think it's a good idea?

peshovp, 2025-05-31:
But if i know my coordinates and i want to put my base station in TIME mode how can i do it?

dremeier, 2025-06-08:
Yes, I think it's a good idea. Because a use case is a temporary base Station for a day or two. At the moment I use the survey-in funktion from the [Missionplanner ](https://github.com/ArduPilot/MissionPlanner)Project.