Stefal/rtkbase #466: F9P detected as u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver when using only "detect"

Stefal, 2025-03-24:
**Describe the bug**
When detecting the receiver from the GUI, a usb connected F9P is detected as `u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver` and the receiver format isn't set

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'SETTINGS'
2. Click on 'DETECT'
3. Accept the result
4. See error

**Expected behavior**
The receiver should be detected as U-Blox_F9P, and receiver format set to `ubx`

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.3
 - OS: Armbian 64bits
 - SBC Orange Pi Zero
 - GNSS Device: Ublox F9P
 - Connection type: usb

**Additional context**
The behavior is ok with `detect&configure`

MikeJWillis, 2025-05-08:
I am having the same issue. 
RTKBase release 2.6.3
OS Ubunut Jammy Server
SBC Orange PI 3B
GNSS: Ublox F9**R** 
Connection: usb

Can detect receiver as above. 
Does not allow configuration .

PavaniVitor, 2025-09-16:
Any news on this? I'm facing the same issue with multiple F9P units.

Stefal, 2025-09-21:
> Any news on this? I'm facing the same issue with multiple F9P units.

With "Detect & Configure" or only with "Detect" ?