Stefal/rtkbase #471: Data logging and RINEX conversion with UM982

paulmteleo, 2025-05-20:
**Describe the bug**
On a fresh install, a UM982 receiver from gnss.store is working as expected.  I am saving logs to the file system with the file rotation set to 2 hours.
I see files accumulating in the specified folder.  Example filenames are:

```
2025-05-20_18-14-53_GNSS-1.rtcm3
2025-05-20_18-14-53_GNSS-1.rtcm3.tag
```

When I view these files, they contain binary data.

Although these files are present in the specified log folder, they are not displayed under the LOG tab of the 
web UI.  The file display area text says **"No matching records found"**

If I zip these files into a .zip file, e.g. test1.zip, the zip file is also not displayed under the LOG tab of the web UI.
I wish to use the RINEX conversion utility to convert a .zip file to RINEX as I usually do with F9P data records.

**To Reproduce**
Steps to reproduce the behavior:
1. install from default installation
2. detect and configure the UM982 receiver. Receiver has connection to a rooftop antenna and shows nominal signal strength on ~20 satellites.  Position converges rapidly to the correct location.
3. Enable file service, setting the file rotation period to 2 hours. Set the data folder to home/paul/rtkbase/um982/data/20250520.
4. Validate that files are being saved to the specified folder (yes, they are being saved!)  Wait for a few hours.
5. zip  *.rtcm and *.rtcm.tag files into  test1.zip in the same folder
6. Open the LOGS tab on the web UI

**Expected behavior**

1. I expect the saved files to be displayed in the LOGS tab of the web UI, but no files are listed. I see `No matching records found`
2. I expect to be able to click the pencil icon in the right column of row with test1.zip to request conversion to RINEX format. test1.zip does not appear in the file list.

**Software/Hardware (please complete the following information):**
```
 - RTKBase release number 2.6.3
 - OS: Ubuntu 20.04.6 LTS
 - SBC IBM Intel I7 thinkpad laptop
 - GNSS Device: Unicore_UM982 - R4.10Build13495
 - Connection type: USB  /dev/ttyUSB0
```

paulmteleo, 2025-05-21:
May I request some explanation: is the behavior I am seeing expected, or is a bug?
If this is expected behavior, can you please explain why this behavior differs from logging with an F9P receiver.
Since my objective is to create a RINEX file from data collected from UM982, can anyone suggest a workaround to achieve my objective.
Big picture. I wish to log data with a mobile UM982 in a location where data comms are absent or unreliable.  I plan to convert the saved data to RINEX format, then submit the RINEX to an online utility such as CSRS to generate the position history in post processing.
Thanks.

Stefal, 2025-05-21:
Hi!
Thanks for your very detailed report. It helps a lot.

What you see is not expected. It think there is a bug.

Could you move some files to the default location `rtkbase/data/` and check the log UI to confirm it works?

paulmteleo, 2025-05-21:
@stefal, many thanks for your rapid response.  Much obliged.
I moved the files to the default location rtkbase/data/  
Now I see them on the LOG tab.  (don't know why I did not think to try that myself!)
I was able to convert to RINEX format and CSRS was able to process the static RINEX data.
For me, that is a big step forward.  Thanks for your assistance.

Stefal, 2025-05-22:
This bug will be fixed in the next release.

commit 42fc7f6e (2025-05-22): Fix #471
RTKLIB instance was created before app.config["DOWNLOAD_FOLDER"] was updated to the correct directory.  Then the logs GUI was not displayed the files in the custom folder.
commit d2b8e08c (2025-07-03): Fix #471
closed 2025-07-03 by Stefal