Stefal/rtkbase #475: No location displayed but plenty satellites available

m-mueller-sl, 2025-06-05:
**Describe the bug**
Not sure whether this is necessarily a bug within this project or a problem with my receiver, but I am in need of some guidance of what's wrong here.

I've setup the tje project on a PI 3b with a Waveshare LC29H BS. The survey-in was done before and worked with an accuracy of 5m.
The pi has been up and running for a few days now, but I still don't get any coordinates or location displayed on the map. The list of satellits does include about 30 to 40 satellites, so enough should be available for acquiring a position.
[image]
The location of my antenna is not optimal. It has some view of the sky but about 50% is obstructed by roof. I am planning on mounting it on the side of the roof for better view, but this will make it less accessible so I'd like to resolve possible problems before.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.3
 - OS: Raspberry Pi Os
 - SBC Raspberry Pi 3B+
 - GNSS Device: LC29H(BS)
 - Connection type: Serial

jaerabr, 2025-06-09:
Hello, how are you?

Check if the output of satellite ephemeris in RTCM format is enabled.

Stefal, 2025-06-09:
>  I still don't get any coordinates or location displayed on the map. 

The realtime location algorithm need L1/L2 frequency and your receiver provides only L1/L5.
It won't work as is, but you lost only the realtime marker, it won't impact the rtcm streams.

Stefal, 2025-06-09:
The simpleRTK2B is L1/L2.

Yes, you need to change the GNSS receiver in your base station to another model. Better choose a L1/L2/L5 receiver. The Unicore UM980 is a pretty good choice.
You need to check the spec of your base station antenna too.

closed 2025-06-09 by Stefal

Stefal, 2025-06-09:
ANN-MB2 is ok for a rover, but not for a base station.

tszabon, 2025-06-16:
To displayed location on the map enable outputting satellite ephemeris in RTCM format, send this command to your LC29H-BS :
$PAIR436,1 ( or $PAIR436,1*26 with checksum).