Stefal/rtkbase #483: Main service off after reboot

Jaxz13, 2025-07-11:
Main service go off after reboot.

Reboot made by crotab every night.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.3
 - OS: Raspberry Pi Os 12
 - SBC Raspberry Pi 3B+,
 - GNSS Device:  Ublox F9P
 - Connection type:  usb

Stefal, 2025-07-23:
Do you have some logs to help me understand this?
`sudo journalctl -u str2str_tcp.service`

Is the main service off too when you do a manual reboot?

Jaxz13, 2025-08-07:
Main service is off very irregularly. Normally  after next manual boot it is up again.

Stefal, 2025-09-15:
If the main service and the ntrip service are both green, I think you have a hardware problem. I would check the F9p -> Raspberry usb cable

closed 2025-09-16 by Stefal