Stefal/rtkbase #488: Dynamic for F9P is't static

Jef239, 2025-08-16:
Add CFG-NAVSPG-DYNMODEL,2 to U-Blox_ZED-F9P_rtkbase.cfg 

By default dynamic is PORTABLE. It's wrong for base station.

Jef239, 2025-08-19:
A little about myself. Experience in RTK development - 20 years, PPP - 10 years. I have my own, proprietary RTK/PPP code. Very roughly - my own RTKLIB. The last 3 years - I have been working at GNSS.STORE. Author of ELT_RTKBase, which is an extension of your rtkbase. https://gnss.store/95-cors-fanless-stations

A small tutorial on GNSS was written, here it is
https://gnss.store/blog/category/gnss-basics.html
https://gnss.store/blog/category/about-gnss-receivers.html

What is interesting to you personally - here are recommendations for setting up Mosaic-X5 and Unicore UM980 in base station mode https://gnss.store/blog/post/optimizing-onocoy-quality-via-receiver-settings.html

Jef239, 2025-09-21:
About the zero baseline double difference test - https://www.comnavtech.com/about/blogs/388.html

I can't provide the source code - it's proprietary. I might be able to provide Windows binaries, but I'll share them with you personally, not post them on GitHub.

It's useless to look at the UM980; when MODE BASE is selected, it always sets the dynamic range to STATIC. You can create an undocumented MODE MOVINGBASE, but I haven't tested whether it allows you to change the dynamic range.

The F9P always uses "moved base" mode. It switches to true baseline mode when setting fixed coordinates and static dynamic range.

We (GNSS.STORE) have quite a few receivers and 4- and 8-channel splitters. So, I'll take some pictures for you on the X20P. And if that's not enough, I'll take some pictures on the F9P.

For now, these are old pictures from the F9P; I don't remember which dynamic range was used.

These are the second differences in the code with a common leading satellite.

[image]

Second рhase differences with a common leading satellite. It's clear that there are intersystem shifts.

[image]

Second phase differences for a common master satellite for GPS and GALILEO. Everything is fine here.

[image]

Second phase differences for GLONASS. Inter-character shifts are visible, that is, shifts between satellites operating on different frequencies.

[image]

Second phase differences for Beidou. A half-cycle shift relative to GPS is visible.

[image]

Second phase differences with a separate leading suppository for each system.

[image]

All of this was taken five years ago or more. So, everything could be much better with the new firmware.

The code scales are in meters, and the phase scales are in cycles.

Well, now you know how RTK receivers are tested.

I'll take fresh measurements; it shouldn't be too difficult for me.

Jef239, 2025-09-24:
Tests on ZED-X20P HPG 2.00 (1e963fM)

STATIC. Code noise is approximately 25 centimeters.
[image]

PORTABLE. The noise level of the code is approximately 75 centimeters. THREE times greater.
[image]

STATIC. Phase noise is approximately 0.01 cycles, or about 2 millimeters.
[image]

PORTABLE. Well, there's a lot going on here. Systematic trends, shifts. The firmware is clearly raw.
[image]

Is this enough for you, or would you like to see it on F9P?

Just in case, here's the raw data.

[portable3_portable9_Code.csv](https://github.com/user-attachments/files/22524291/portable3_portable9_Code.csv)
[portable3_portable9_Phase.csv](https://github.com/user-attachments/files/22524292/portable3_portable9_Phase.csv)
[static3_static9_Code.csv](https://github.com/user-attachments/files/22524294/static3_static9_Code.csv)
[static3_static9_Phase.csv](https://github.com/user-attachments/files/22524297/static3_static9_Phase.csv)

Jef239, 2025-09-29:
F9P 1.51

STATIC CODE
[image]

PORTABLE CODE
[image]

STATIC PHASE
[image]

PORTABLE PHASE
[image]

The differences between STATIC and PORTABLE in the F9P aren't as significant as in the X20P. However, STATIC mode clearly has lower code and phase noise, meaning better quality. It's not dramatic, but it's about 20% better.

Stefal, 2025-10-18:
Thank you @Jef239 for your extensive report.
I'll add the static setting in the F9P configuration. And perhaps request the end user to launch the automatic configuration after the next RTKbase update.

Do you think we can repro your tests with tools like Anubis, or Ringo ?