Stefal/rtkbase #499: Debian 12 (bookworm) with Mosaic X5 via USB - system hangs after activating SBF stream (configure)

BodoBookhagen, 2025-09-21:
**Describe the bug**
I am running into issues when trying to setup RTKBase v2.6.3 on Raspberry Pi OS (64-bit) Debian 12 (bookworm) with a mosaic-X5 connected via USB. I am asking for help to get the SBF stream from the X5 to the RPi to work. Please note that the same hardware is working fine with RTKBase v2.6.3 on Debian 11 (bullseye). I will need to migrate to Debian 12 to continue receiving updates and make sure the systems will run in the years to come. I also have tried the image provided by the link in the installation section of RTKBase.

The RPi and X5 works fine until the setup of the SBF stream and starting the RTKBase services. I can log to the SD card on the X5 and download the data. But SBF/NMEA streaming causes the RPi to hang (continuous green light, no response, also when unplugging the X5 again). If I plug in the X5 in a different computer and run the setup to delete the stream (while keeping all other settings), the RPi is working again. However, gpsd/chronyd depends on the SBF output via USB and I am interested to have this working. The casting works, when starting the services and configuring the device (on a clean system)- but upon reboot the system hangs (wont' come up at all and no ssh login possible.)

I can read out the NMEA/SBF from the same Mosaic-X5 device on an Intel/AMD ubuntu, but not on a RPi Debian 12. I realize that this may not be linked with RTKBase, but I really would like to run RTKBase on Debian 12.

Detailed steps taken:

```bash
pi@qt17:~ $ sudo ./install.sh -e
user for RTKBase is:  pi
################################
USB GNSS RECEIVER DETECTION
################################
/dev/ttyGNSS  -  Septentrio_Septentrio_USB_Device_3875036 -  115200 -  4.14.4
GNSS RECEIVER DETECTED: /dev/ttyGNSS  -  Septentrio_Septentrio_USB_Device_3875036  -  115200
pi@qt17:~ $ sudo ./install.sh -c
user for RTKBase is:  pi
################################
CONFIGURE GNSS RECEIVER
################################
Mosaic-X5 Firmware:  4.14.4
Resetting the mosaic-X5 settings....
Connection closed
Waiting 30s for mosaic-X5 reboot
Sending settings....
Failed...retrying in 2s
Failed...retrying in 2s
Failed...retrying in 2s
Septentrio Mosaic-X5 successfuly configured
```
Now starting the services will work, and casting is running. But upon reboot, the X5 is detected (see below), but the system hangs (continuous green light, no further entry in journalctl)

```
Sep 21 13:18:31 qt17 kernel: usb 1-1.1.2: new high-speed USB device number 6 using dwc_otg
Sep 21 13:18:31 qt17 kernel: usb 1-1.1.2: New USB device found, idVendor=152a, idProduct=85c0, bcdDevice= 3.14
Sep 21 13:18:31 qt17 kernel: usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 21 13:18:31 qt17 kernel: usb 1-1.1.2: Product: Septentrio USB Device
Sep 21 13:18:31 qt17 kernel: usb 1-1.1.2: Manufacturer: Septentrio
Sep 21 13:18:31 qt17 kernel: usb 1-1.1.2: SerialNumber: 3875036
Sep 21 13:18:31 qt17 kernel: cdc_acm 1-1.1.2:1.2: ttyACM1: USB ACM device
Sep 21 13:18:31 qt17 kernel: cdc_acm 1-1.1.2:1.4: ttyACM2: USB ACM device
Sep 21 13:18:31 qt17 kernel: usb-storage 1-1.1.2:1.6: USB Mass Storage device detected
Sep 21 13:18:31 qt17 kernel: scsi host0: usb-storage 1-1.1.2:1.6
Sep 21 13:18:31 qt17 mtp-probe[631]: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2"
Sep 21 13:18:31 qt17 mtp-probe[631]: bus: 1, device: 6 was not an MTP device
Sep 21 13:18:31 qt17 kernel: usbcore: registered new interface driver cdc_ether
Sep 21 13:18:31 qt17 kernel: rndis_host 1-1.1.2:1.0 usb0: register 'rndis_host' at usb-3f980000.usb-1.1.2, RNDIS device, 1a:32:02:99:15:45
Sep 21 13:18:31 qt17 kernel: usbcore: registered new interface driver rndis_host
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.3206] manager: (usb0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/5)
Sep 21 13:18:31 qt17 mtp-probe[638]: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2"
Sep 21 13:18:31 qt17 mtp-probe[638]: bus: 1, device: 6 was not an MTP device
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.3867] device (usb0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: '>
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.3937] settings: (usb0): created default wired connection 'Wired connection 2'
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.3952] device (usb0): carrier: link connected
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.3962] device (usb0): state change: unavailable -> disconnected (reason 'carrier-changed', sys-ifa>
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.3983] policy: auto-activating connection 'Wired connection 2' (da9a2d39-33b5-3c74-a521-04e0ea5d8e>
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.4001] device (usb0): Activation: starting connection 'Wired connection 2' (da9a2d39-33b5-3c74-a52>
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.4005] device (usb0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'mana>
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.4019] device (usb0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.4048] device (usb0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.4061] dhcp4 (usb0): activation: beginning transaction (timeout in 45 seconds)
Sep 21 13:18:31 qt17 avahi-daemon[379]: Joining mDNS multicast group on interface usb0.IPv6 with address fe80::ff30:af29:f20e:8477.
Sep 21 13:18:31 qt17 avahi-daemon[379]: New relevant interface usb0.IPv6 for mDNS.
Sep 21 13:18:31 qt17 avahi-daemon[379]: Registering new address record for fe80::ff30:af29:f20e:8477 on usb0.*.
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.7923] dhcp6 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.7946] policy: set 'Silvio' (wlan0) as default for IPv6 routing and DNS
Sep 21 13:18:31 qt17 avahi-daemon[379]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::19a0:9729:351a:58ab.
Sep 21 13:18:31 qt17 NetworkManager[479]: <info>  [1758457111.8111] dhcp6 (wlan0): state changed new lease
Sep 21 13:18:31 qt17 avahi-daemon[379]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2a02:8109:b518:db00:8fbd:a05e:ca04:cae6.
Sep 21 13:18:31 qt17 avahi-daemon[379]: Registering new address record for 2a02:8109:b518:db00:8fbd:a05e:ca04:cae6 on wlan0.*.
Sep 21 13:18:31 qt17 avahi-daemon[379]: Withdrawing address record for fe80::19a0:9729:351a:58ab on wlan0.
Sep 21 13:18:32 qt17 kernel: scsi 0:0:0:0: CD-ROM            Linux    File-CD Gadget   0314 PQ: 0 ANSI: 2
Sep 21 13:18:32 qt17 kernel: scsi 0:0:0:1: Direct-Access     Linux    File-CD Gadget   0314 PQ: 0 ANSI: 2
Sep 21 13:18:32 qt17 kernel: sd 0:0:0:1: Power-on or device reset occurred
Sep 21 13:18:32 qt17 kernel: sd 0:0:0:1: [sda] Media removed, stopped polling
Sep 21 13:18:32 qt17 kernel: sd 0:0:0:1: [sda] Attached SCSI removable disk
Sep 21 13:18:32 qt17 kernel: scsi 0:0:0:0: Attached scsi generic sg0 type 5
Sep 21 13:18:32 qt17 kernel: sd 0:0:0:1: Attached scsi generic sg1 type 0
Sep 21 13:18:32 qt17 kernel: sr 0:0:0:0: Power-on or device reset occurred
Sep 21 13:18:32 qt17 kernel: sr 0:0:0:0: [sr0] scsi-1 drive
Sep 21 13:18:32 qt17 kernel: cdrom: Uniform CD-ROM driver Revision: 3.20
Sep 21 13:18:32 qt17 kernel: sr 0:0:0:0: Attached scsi CD-ROM sr0
```

I have been trying various things and have used various Raspberry Pi and Mosiac-X5 sensors. The issue persists only with Debian 12 (Debian 11 works fine). Any suggestions or help how to properly setup and connect an X5 via USB that allows casting and continues to work after unexpected reboots would be highly appreciated. 

**Expected behavior**
X5 should work fine after a reboot.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.6.3
 - OS: Raspberry Pi OS (64-bit) Debian 12 (bookworm)
 - SBC  Raspberry Pi 3 Model B+
 - GNSS Device: Mosaic-X5 with Firmware:  4.14.4 
 - Connection type: usb

Stefal, 2025-09-21:
I've encountered this problem before. I thought it was a power supply issue, but your tests with Debian 11 show that the problem may lie elsewhere.
Would you be able to repeat the test with Debian 12, but with a powered USB hub between the Raspberry Pi and the Mosaic-X5?

BodoBookhagen, 2025-09-22:
Thanks for this suggestion. I just tried the same setup with a powered USB hub - the result is the same. The X5 causes the RPi to hang when an SBF/NMEA stream is activated. As soon as I turn off the stream, it works fine.
I am currently logging to the SD Card on the X5 and rsync the data every hour to the RPi. This works well, but I don't run the chronyd/gpsd (which is useful for remote areas without internet connection).

ALSO: I just tried with an F9P (Firmware 1.32) and it works fine. The casting runs and there is no continuous green light. This works without an USB hub. So, the problem is specific to the X5 (which is using a SBF stream and not UBX).

Do you have any other suggestion what to try or turn off to make the casting and SBF logging connection work?